mu-editor/mu #1541: Crash Report

AbouTheGreat, 2021-05-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
[mu.log](https://github.com/mu-editor/mu/files/6434739/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6434742/mu.log)

Traceback (most recent call last):
  File "C:\Users\Abou Ilboudo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Abou Ilboudo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Abou Ilboudo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Abou Ilboudo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\Abou Ilboudo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

tjguk, 2021-05-06:
Thanks for thre report @AbouTheGreat  -- as far as I can tell, the two logfiles you uploaded are identical, so I'll just take one of them. (If you think they should be different in some way, let me know).

This looks like a problem we're already tracking at #1532 so I'll close this issue and reference it over there along with the logs.

closed 2021-05-06 by tjguk

AbouTheGreat, 2021-05-06:
Yes, I didn't mean to upload it twice, thanks

AbouTheGreat, 2021-05-06:
Mu-Editor/Mu is working again, only problem is it's not detecting my
circuit, no matter how many times I reset and update the circuit it
wont detect it.