mu-editor/mu #1542: Crash Report

me-shiv, 2021-05-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64

```
Traceback (most recent call last):
  File "/home/shivansh/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 372, in on_kernel_start
    self.view.add_jupyter_repl(kernel_manager, kernel_client)
  File "/home/shivansh/.local/lib/python3.7/site-packages/mu/interface/main.py", line 581, in add_jupyter_repl
    kernel_client.start_channels()
  File "/home/shivansh/.local/lib/python3.7/site-packages/qtconsole/kernel_mixins.py", line 47, in start_channels
    super(QtKernelClientMixin, self).start_channels(*args, **kw)
  File "/home/shivansh/.local/lib/python3.7/site-packages/jupyter_client/threaded.py", line 248, in start_channels
    super().start_channels(shell, iopub, stdin, hb, control)
  File "/home/shivansh/.local/lib/python3.7/site-packages/jupyter_client/client.py", line 295, in start_channels
    self.hb_channel.start()
  File "/home/shivansh/.local/lib/python3.7/site-packages/jupyter_client/client.py", line 364, in hb_channel
    self._hb_channel = self.hb_channel_class(self.context, self.session, url)
  File "/home/shivansh/.local/lib/python3.7/site-packages/qtconsole/util.py", line 65, in __new__
    QClass.__init__(inst)
TypeError: __init__() missing 1 required positional argument: 'context'

```

carlosperate, 2021-05-06:
Thanks for the report @me-shiv!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

This looks similar to https://github.com/mu-editor/mu/issues/1444, could you also run `pip3 list` (or whichever pip you've used to install Mu) in a terminal and paste the output here?

carlosperate, 2021-06-04:
Hi @me-shiv, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `~/.cache/mu/`.

carlosperate, 2021-06-23:
Hi @me-shiv,

If you are still experiencing this issue could you try running this command in a terminal and trying to launch Mu again? Hopefully this should resolve it:

```
pip3 install jupyter-client==6.1.12
```

carlosperate, 2021-08-10:
Hi @me-shiv,

The latest beta might have resolved this issue. Would you be able to check if it works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5
`pip install mu-editor==1.1.0b5`

Thanks!

carlosperate, 2021-09-24:
Hi @me-shiv, just wondering if you had a chance to test this with beta.5?

carlosperate, 2021-12-17:
@me-shiv, I think this crash might be related to issue: #1831

We think we might have found a solution to that problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.

carlosperate, 2022-05-17:
Closing due to lack of activity.

closed 2022-05-17 by carlosperate