mu-editor/mu #1543: M1 Mac issue with adafruit-board-toolkit

dhalbert, 2021-05-07:
We are seeing an issue that MacOS on M1 Macs is not returning the USB CDC interface names properly to `adafruit_board_toolkit`, so it can't detect that a serial port is on a CircuitPython.

It works fine on non-M1 Macs up through MacOS 11.3.1, but is not working on M1 Macs on 11.3.1 and earlier ones. We're attempting to do more testing, but I don't personally have an M1 Mac and neither does @Neradoc, who wrote the code to properly get the interface names.

This would explain some issues being reported around CircuitPython boards not being detected on MacOS in the beta.,

carlosperate, 2021-05-07:
Thanks for letting us know!

In the affected cases, does it happen all the time, or can it happen sporadically as well? People have reported similar issues in the Gitter channel, but in some cases it works at the beginning and then stops working, could that be related?
https://gitter.im/mu-editor/general

dhalbert, 2021-05-07:
It seems to just not detect a CircuitPython board. I don't think we've seen it sometimes working. The gitter problems also include some weird packaging or path issues where the wrong version of pyserial was loaded. That causes a crash. This is not a crash.

The problem is that the USB interface name for the serial port is not what it should be (it may just be empty?). `adafruit-board-toolkit` filters the ports based on the interface name, looking for "CircuitPython CDC " as a prefix on the interface name (and some other prefixes for non-Adafruit boards). It's not finding any names that match.

A simple test on a Mac is to do this:
```py
# unplug all CircuitPython boards

$ pip3 install --upgrade adafruit-board-toolkit   # Make sure it's up to date

$ python3

>>> import adafruit_board_toolkit.circuitpython_serial
>>> print(adafruit_board_toolkit.circuitpython_serial.comports())

# now plug in a CircuitPython board and wait for CIRCUITPY to appear

>>> print(adafruit_board_toolkit.circuitpython_serial.comports())
```
On a non-M1 Mac, this will show ports, on M1 it will not. `adafruit-board-toolkit` it doing something like this internally:
```py
>>> import serial.tools.list_ports
>>> print(serial.tools.list_ports.comports())
```
And then it is further filtering the returned comports by interface name:
https://github.com/adafruit/Adafruit_Board_Toolkit/blob/03d44d36e12e0cf27de1f9b4db9f2093be6a5f25/adafruit_board_toolkit/circuitpython_serial.py
```py
    return tuple(
        port
        for port in ports
        if port.interface and port.interface.startswith("CircuitPython CDC")
    )
```
We have someone debugging this by looking at `ioreg -l` with a board plugged in. There is discussion in the Adafruit Discord #circuitpython-dev channel from yesterday evening where we're discussing this.

ntoll, 2021-05-07:
@dhalbert thanks for the heads up. If you need additional testing bandwidth, my youngest has an M1 Macbook I could borrow for such purposes.

Neradoc, 2021-05-07:
The code finds every serial ports as `IOSerialBSDClient` in the IO registry and then goes up to find the parent `IOUSBInterface`, by class name as reported by `IOObjectGetClass()`, which holds the interface name as a `USB Interface Name` property.

On M1 the interface object is now reported by `IOObjectGetClass()` as a `IOUSBHostInterface`. It already was that class, but had a property: `"IOClassNameOverride" = "IOUSBInterface"` that is now gone. [Even though `IOUSBHostInterface` is also marked as deprecated, but later](https://developer.apple.com/documentation/kernel/iousbhostinterface).
So
- look for `IOUSBHostInterface`
- if not found, look for `IOUSBInterface`

But that is not all.
On M1 the `USB Interface Name` property is gone. But there is a `kUSBString` property with the same content. `kUSBString` was already there, I don't know if it came up at some point and `USB Interface Name` was deprecated in favor of it or what.
Looking up `kUSBString` on Apple's developer site, or Google yields nothing... *"Did you mean SubString ?"* 🙄.

So I believe this version works:
https://gist.github.com/Neradoc/e718ee1db78549e510dfd03da2eed7d9

Hugo tested it on the discord, I don't mind more tests though, see if "CircuitPython CDC data" comes up somewhere when running the script, like that:
```sh
python3 pyserial_list_ports_osx.py
```
> /dev/cu.usbmodem144443301: QT Py M0 Haxpress - CircuitPython CDC data [USB VID:PID=239A:00CC SER=6EC079803935545020312E321E200FFF LOCATION=20-4.4.4.3.3]

I'll PR it to `Adafruit_Board_Toolkit`. And pyserial if anybody cares over there.

dhalbert, 2021-05-10:
I probably can't work on this in earnest until Tuesday evening or Wednesday but it is a first priority for me then. On the Mu side it is just a matter of bumping the required `adafruit-board-toolkit` version after I make a release that incorporate's @Neradoc's fixes.

ntoll, 2021-05-10:
@dhalbert yup... totally understand WRT Mu just needing to bump the version. But always happy to help if need be for testing etc... :+1:

carlosperate, 2021-05-12:
Currently Mu supports macOS 10.12 Sierra, the cut off really being driven by Qt and we are staying in slightly older versions to maximise OS compatibility.

Can the devices still be detected via VID even if we don't get as much information in older systems?

dhalbert, 2021-05-12:
> Can the devices still be detected via VID even if we don't get as much information in older systems?

Yes, there can be fallbacks, and these can be in `adafruit-board-toolkit`, instead of in Mu, so as new kinds of boards show up, we would just update the library. So the version of `adafruit-board-toolkit` should not be pinned to a single version, but should be restricted to, say, `<2.0.0`, or maybe `~=1.0.2`.

CircuitPython now allows the REPL serial channel to be turned off dynamically. Would it be a goal of Mu to try to recognize the board as a CircuitPython board even if there is no serial available? That would be done with VID, VID/PID, or by the presence of a CIRCUITPY drive. But the serial button would not work.

ntoll, 2021-05-13:
Hmm... not sure about Mu goal and CP boards that have no serial. My hunch is to support as much stuff as possible, although this would add complexity (enabling/disabling serial button etc...). I assume it'd be easy to use the `adafruit-board-toolkit` to work out this status for an attached board..?

dhalbert, 2021-05-13:
In Mu 1.0.x and 1.1.0-beta.3 right now there has to be a serial port available for Mu to recognize the board. 1.0x uses the VID/PID info from the port to auto-choose the mode. So it's not a regression not to support boards with no serial. Eventually it could be supported, but I'm OK with deferring that for now.

ntoll, 2021-05-13:
Ack. Happy to be guided by you on this one. Please never hesitate to say, "Hey CircuitPython / Adafruit boards do such-and-such, you should think about supporting that"..! Always fun to make fun stuff happen in CircuitPython mode. :-)

dhalbert, 2021-05-19:
We have a fix for the M1 Mac issue that has been incorporated into adafruit-board-toolkit 1.1.0. I tested adafruit-board-toolkit on an M1 Mac and it works.

However, I'm having trouble testing this in a Mu installation, installed with the .dmg installer. I'm confused as to the purpose of the venv that's in, say, `/Users/halbert/Library/Application Support/mu/mu_venv-38-20210513-135550`. That venv contains a copy of `adafruit-board-toolkit`. I tried updating that `adafruit-board-toolkit` to 1.1.0 there, to no avail. However, there is also a 1.0.1 version in `/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages`, and I think that is the one I should be updating to test.

The lib in the `site-packages` directory are what Mu uses to run with, right? That appears to be true based on the Python path listed in the logs. What is the venv used for? I don't see a reason for it to contain `adafruit-board-toolkit`. Is this about the `mu.wheels` dependencies versus `setup.py` dependencies ([discussion](https://github.com/mu-editor/mu/pull/1371#issuecomment-809696953)?

Is there a standard way you test updating an internally-needed package?

carlosperate, 2021-05-19:
Sorry about the confusion, I've just merged https://github.com/mu-editor/mu/pull/1491, which removes the dependency from `mu.wheels` as it wasn't needed there.

The easiest way to test it would be to create a PR with the updated dependency and the CI will create a dmg.

If you need to test locally before the dependency it's available in PyPI then you can probably `pip install -e` using the bundle python:

```
mu-editor.app/Contents/Resources/Python/bin/python3.8 -m pip install -e ....
```

PR #1580 (2021-05-20): Update adafruit-board-toolkit to ~=1.1 to fix M1 MacOS issues
Updates the version specifier for `adafruit-board-toolkit` so the minimum version is `1.1.0.` Any 1.x version will be upward compatible and be OK.

`adafruit-board-toolkit` 1.1.0 has two important fixes:
- MacOS interface name fetching is improved so it works back to Sierra and forward to MacOS Big Sur on M1. There are differences between M1 MacOS and non-M1. See https://github.com/adafruit/Adafruit_Board_Toolkit/pull/4.
- The interface names don't have to start with "CircuitPython"; there are some boards that uses other prefixes. S [... cut]
closed 2021-05-20 by ntoll