mu-editor/mu #1550: Crash Report

jdimpson, 2021-05-09:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64

```
Traceback (most recent call last):
  File "/home/user1/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/home/user1/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 138, in download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user1/.local/lib/python3.7/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/user1/.local/lib/python3.7/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/user1/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/user1/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/home/user1/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download esptool==3.*

```

jdimpson, 2021-05-09:
[mu.log](https://github.com/mu-editor/mu/files/6446828/mu.log)

carlosperate, 2021-05-11:
Thank you very much for the crash report and specially the log @jdimpson!

What version of `pip` and `setuptools` do you have in the Python environment you've used to install the mu-editor?

The interesting part of the log:
```
2021-05-08 21:58:48,314 - mu.virtual_environment:113(download) INFO: Running pip download for esptool / esptool==3.* / []
2021-05-08 21:58:49,426 - mu.virtual_environment:129(download) DEBUG: Collecting esptool==3.*
  Using cached https://files.pythonhosted.org/packages/dd/3d/d1d4c004927e6e6807c441ce70330ed969c725d2906053fbd2ff994b4439/esptool-3.0.tar.gz
  Saved /home/user1/.local/lib/python3.7/site-packages/mu/wheels/esptool-3.0.tar.gz
    Complete output from command python setup.py egg_info:
    running egg_info
    creating pip-egg-info/esptool.egg-info
    writing pip-egg-info/esptool.egg-info/PKG-INFO
    writing dependency_links to pip-egg-info/esptool.egg-info/dependency_links.txt
    writing requirements to pip-egg-info/esptool.egg-info/requires.txt
    writing top-level names to pip-egg-info/esptool.egg-info/top_level.txt
    writing manifest file 'pip-egg-info/esptool.egg-info/SOURCES.txt'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-download-ha99juts/esptool/setup.py", line 122, in <module>
        entry_points=entry_points,
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/egg_info.py", line 296, in run
        self.find_sources()
      File "/usr/lib/python3/dist-packages/setuptools/command/egg_info.py", line 303, in find_sources
        mm.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/egg_info.py", line 534, in run
        self.add_defaults()
      File "/usr/lib/python3/dist-packages/setuptools/command/egg_info.py", line 574, in add_defaults
        rcfiles = list(walk_revctrl())
      File "/usr/lib/python3/dist-packages/setuptools/command/sdist.py", line 20, in walk_revctrl
        for item in ep.load()(dirname):
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2410, in load
        self.require(*args, **kwargs)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2433, in require
        items = working_set.resolve(reqs, env, installer, extras=self.extras)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 791, in resolve
        raise VersionConflict(dist, req).with_context(dependent_req)
    pkg_resources.VersionConflict: (setuptools 40.8.0 (/usr/lib/python3/dist-packages), Requirement.parse('setuptools>=45'))
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-download-ha99juts/esptool/
```

It sounds like `esptool` has a requirement for `'setuptools>=45'`, but I couldn't find anything in the source code that would indicate that: https://github.com/espressif/esptool/blob/v3.0/setup.py
In fact, earlier in the log it downloads the wheel for v56.1.0 from dependency `ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.7.4` which only needs `setuptools>=18.5`:
```
Collecting setuptools>=18.5 (from ipython>=5.0.0->ipykernel>=4.1->qtconsole==4.7.4)
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/11/1b/7012b145cb228aed20f9b2b8b259df49e7963d900799ea44791f54d06ab9/setuptools-56.1.0-py3-none-any.whl
  Saved /home/user1/.local/lib/python3.7/site-packages/mu/wheels/setuptools-56.1.0-py3-none-any.whl
```

carlosperate, 2021-08-10:
Hi @jdimpson,

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5
`pip install mu-editor==1.1.0b5`

Thanks!

jdimpson, 2021-09-21:
Oh my goodness I'm so sorry for forgetting about this. The hard drive died on the system that had this error. I replaced it and in the process I installed the latest Debian, and have had no issues installing and running mu-editor.

Perhaps not a satisfying resolution, especially if someone else has the same problem, but upgrading to the latest Debian then installing mu-editor via pip3 solved my problem.

Thank you for following up, and apologies again for letting this issue flounder.

carlosperate, 2021-09-21:
No worries, thanks for the update!

It's quite possible the changes in beta.5 and using a more recent version of `setuptools` and `pip` resolved the problem. Hopefully is somebody encounters this issue looking for a solution they can try updating these.

I'll close this as resolved, thanks @jdimpson!

closed 2021-09-21 by carlosperate