mu-editor/mu #1553: Crash Report

peelu1, 2021-05-10:
[mu.txt](https://github.com/mu-editor/mu/files/6449643/mu.txt)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_IN
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-05-10:
Thanks for the report @peelu1, and for attaching the log file!

The interesting bit from the log:

- First we get a `FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\idea\\AppData\\Local\\python\\mu\\mu_venv-38-20210509-222823\\Lib\\site-packages'`, which then causes a `PermissionError: [WinError 5] Access is denied:` when trying to rename the venv to quarantine.
```
2021-05-10 10:24:16,036 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\idea\\AppData\\Local\\python\\mu\\mu_venv-38-20210509-222823\\Lib\\site-packages'
2021-05-10 10:24:16,239 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\idea\AppData\Local\python\mu\mu_venv-38-20210509-222823 as C:\Users\idea\AppData\Local\python\mu\mu_venv-38-20210509-222823.FAILED
Traceback (most recent call last):
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\idea\AppData\Local\python\mu\mu_venv-38-20210509-222823

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\idea\\AppData\\Local\\python\\mu\\mu_venv-38-20210509-222823' -> 'C:\\Users\\idea\\AppData\\Local\\python\\mu\\mu_venv-38-20210509-222823.FAILED'
```

- Then it's unable to upgrade pip?

```
2021-05-10 10:24:32,817 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\idea\AppData\Local\python\mu\mu_venv-38-20210509-222823\scripts\python.exe exists
2021-05-10 10:24:55,754 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 3221225786; output: Requirement already satisfied: pip in c:\users\idea\appdata\local\python\mu\mu_venv-38-20210510-102416\lib\site-packages (21.0.1)
Collecting pip
  Using cached pip-21.1.1-py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.0.1
    Uninstalling pip-21.0.1:
      Successfully uninstalled pip-21.0.1
2021-05-10 10:24:55,895 - mu.virtual_environment:468(quarantine_failed_venv) INFO: Quarantined C:\Users\idea\AppData\Local\python\mu\mu_venv-38-20210510-102416 as C:\Users\idea\AppData\Local\python\mu\mu_venv-38-20210510-102416.FAILED
2021-05-10 10:24:55,895 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x000002970C6D7790>; Splash Handler; <SplashLogHandler (INFO)>
2021-05-10 10:25:03,205 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\idea\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
```
- And then (I guess in the next venv attempt) when it tries to install the packages, it looks like it fails at some point but it doesn't print any errors and it doesn't capture any exceptions? Basically this bit lists all the packages it aims to install, but the pip list afterways only got half way through:
```
2021-05-10 10:25:24,559 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
2021-05-10 10:28:28,224 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\idea\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl

Processing c:\users\idea\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl

Processing c:\users\idea\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl

Processing c:\users\idea\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl

Processing c:\users\idea\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl

[... cut ...]
python-dateutil  2.8.1

pywin32          300

pyzmq            22.0.3

setuptools       56.0.0

six              1.15.0

tornado          6.1

traitlets        5.0.5

wcwidth          0.2.5

wheel            0.36.2
2021-05-10 10:28:34,691 - mu.virtual_environment:680(install_jupyter_kernel) INFO: Installing Jupyter Kernel: mu_venv-38-20210509-222823
```
- So, then `self.ensure()` fails to fing `pygamezero`, tries again and encounter the same situation.

There is quite a few things going on this one...

carlosperate, 2021-07-05:
Hi @peelu1! 
We've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-08-10:
Hi @peelu1,

If you'd be able to check if beta.5 works that's be really appreciated. Could you give it a run and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@peelu1 have you been able to test beta.5? There has been some fixes in this area that could have resolved the problem you were experiencing: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate