mu-editor/mu #1558: Crash Report - fresh install on Ubuntu 20.04 LTS

abixadamj, 2021-05-12:
[mu.log](https://github.com/mu-editor/mu/files/6463873/mu.log)
I just tried to install fresh Mu, because I have a webinar today - and after install all needed stuff.....

**REMEMBER TO ATTACH YOUR LOG FILE - YES - I remember ;-)**

```
Version: 1.1.0.beta.4
Locale: pl_PL
Platform: Linux 5.8.0-50-generic #56~20.04.1-Ubuntu SMP Mon Apr 12 21:46:35 UTC 2021 x86_64

adasiek@adasiek-Latitude-E7240:~/mu$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

```

```
Traceback (most recent call last):
  File "/home/adasiek/mu/mu/app.py", line 161, in run
    raise ex
  File "/home/adasiek/mu/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 491, in ensure_and_create
    self.create()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 620, in create
    self.create_venv()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 651, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/adasiek/.local/share/mu/mu_venv-38-20210512-082344

```

abixadamj, 2021-05-12:
I saw, that there is no `/home/adasiek/.local/share/mu` directory, so I created it manually and then tried to run another time:

```
 File "/home/adasiek/mu/mu/virtual_environment.py", line 651, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/adasiek/.local/share/mu/mu_venv-38-20210512-082808
2021-05-12 08:28:16,211 - mu.settings:169(save) DEBUG: Saving to /home/adasiek/.local/share/mu/session.json
2021-05-12 08:28:16,211 - mu.settings:169(save) DEBUG: Saving to /home/adasiek/.local/share/mu/venv.json
QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
QSocketNotifier: Invalid socket 17 and type 'Exception', disabling...
Naruszenie ochrony pamięci (zrzut pamięci)
```
The last is Memory dump error......
But in file: Saving to /home/adasiek/.local/share/mu/venv.json there is:
```
{
  "dirpath": "/home/adasiek/.local/share/mu/mu_venv-38-20210512-082808"
}

```
But there is NO such directory.....
@ntoll - maybe you can help..

abixadamj, 2021-05-12:
I got it !
I need to install `python3-virtualenv`, but this I read somewhere inside a log.

[image]

I will open another Issue with my thoughts about that.

closed 2021-05-12 by abixadamj

carlosperate, 2021-05-12:
What steps did you run to install Mu?
virtualenv is part of the `install_requires` so it should have been installed automatically when Mu was pip installed? Or at least an error should have been thrown if the OS blocked the installation somehow? (Debian probably prefers the virtualenv module to be apt installed instead?).