mu-editor/mu #1559: Crash Report

Alice-star-lgtm, 2021-05-12:
[mu.log](https://github.com/mu-editor/mu/files/6464527/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6464533/mu.log)

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_GB
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 283, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 626, in create
    self.register_baseline_packages()
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 751, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 289, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

carlosperate, 2021-05-12:
Thanks for the report, and specially for attaching the logs, they have been really useful!

carlosperate, 2021-05-12:
## First issue: Installing packaged wheels

It starts by creating a new venv:

```
2021-05-12 08:33:37,312 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-12 08:33:37,312 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-083249
2021-05-12 08:33:37,312 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210512-083249
2021-05-12 08:34:38,259 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-12 08:34:38,259 - mu.virtual_environment:648(create_venv) INFO: Created virtual environment using C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-083249
```

Looks like it fails to install some of the pacakged wheels, but it doesn't throw/log an error:
```
2021-05-12 08:36:36,954 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
2021-05-12 08:39:39,846 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl

Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cryptography-3.4.7-cp36-abi3-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\decorator-5.0.7-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ecdsa-0.16.1-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\esptool-3.0-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\flask-1.1.2-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipykernel-5.5.3-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython-7.22.0-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython_genutils-0.2.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\itsdangerous-1.1.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jedi-0.18.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jinja2-2.11.3-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_client-6.1.12-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_core-4.7.1-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\markupsafe-1.1.1-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\numpy-1.20.2-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\parso-0.8.2-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pgzero-1.2.1-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pickleshare-0.7.5-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\prompt_toolkit-3.0.18-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pycparser-2.20-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pygame-2.0.1-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pygments-2.8.1-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pyserial-3.5-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\python_dateutil-2.8.1-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pywin32-300-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pyzmq-22.0.3-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\qtconsole-4.7.4-py2.py3-none-any.whl
[... cut ...]
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\alice\\AppData\\Local\\python\\mu\\mu_venv-38-20210512-084040' -> 'C:\\Users\\alice\\AppData\\Local\\python\\mu\\mu_venv-38-20210512-084040.FAILED'
```

Then it set ups the next venv and the same issue repeats 3 times until it runs out of attempts.

carlosperate, 2021-05-12:
@Alice-star-lgtm do you have any antivirus software running on the computer?
I think @tjguk mentioned that in some cases it can lock files/directories as Mu is trying to create them and that could explain some of the errors we are seeing in the log.

If you do have an antivirus software running, would it be possible to temporarily disable and then try to open Mu?
If you could then add the latest version of the log file here it would really help us see if that was a trigger for this problem, so that we can update Mu to be able to work around it.

Alice-star-lgtm, 2021-05-12:
Hi I have Mcafee Live Safe running. I can turn off Real Time Scanning and Firewall. I don't have an option to just 'disable'

Alice-star-lgtm, 2021-05-12:
Same issue, real time scanning and firewall off Bug report #1562

carlosperate, 2021-05-12:
Merging https://github.com/mu-editor/mu/issues/1562 here:

> 
> [mu.log](https://github.com/mu-editor/mu/files/6465785/mu.log)
> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_GB
> Platform: Windows 10 10.0.18362 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
>     self.ensure()
>   File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
>     self.ensure_key_modules()
>   File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
>     raise VirtualEnvironmentEnsureError(
> mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
> 
> ```

carlosperate, 2021-05-12:
Thank you very much for the extra testing @Alice-star-lgtm, this has been indeed very useful!

Looks like the secondary issues failing to quarantine the old venv, and the issues trying to create new ones (basically the 2nrd and 3rd sections in https://github.com/mu-editor/mu/issues/1559#issuecomment-839650490) do no happen when the antivirus is disabled.
<details>
<summary>Click here to see the log</summary>
<pre><code>
-----------------

Starting Mu 1.1.0.beta.4
2021-05-12 13:14:28,370 - root:266(run) INFO: uname_result(system='Windows', node='LAPTOP-BSE36TIV', release='10', version='10.0.18362', machine='AMD64', processor='AMD64 Family 21 Model 112 Stepping 0, AuthenticAMD')
2021-05-12 13:14:28,370 - root:266(run) INFO: uname_result(system='Windows', node='LAPTOP-BSE36TIV', release='10', version='10.0.18362', machine='AMD64', processor='AMD64 Family 21 Model 112 Stepping 0, AuthenticAMD')
2021-05-12 13:14:28,370 - root:267(run) INFO: Python path: ['C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-12 13:14:28,370 - root:267(run) INFO: Python path: ['C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\alice\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-12 13:14:28,370 - root:268(run) INFO: Language code: en_GB
2021-05-12 13:14:28,370 - root:268(run) INFO: Language code: en_GB
2021-05-12 13:14:28,371 - mu.settings:220(load) WARNING: No settings file found at C:\Users\alice\AppData\Local\python\mu\settings.json; skipping
2021-05-12 13:14:29,986 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-12 13:14:29,986 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-12 13:14:36,765 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-12 13:14:36,770 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-100025
2021-05-12 13:14:36,807 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-12 13:14:36,808 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-100025
2021-05-12 13:14:36,808 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210512-100025
2021-05-12 13:14:36,826 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210512-100025
2021-05-12 13:14:39,904 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\alice\\AppData\\Local\\python\\mu\\mu_venv-38-20210512-100025\\Lib\\site-packages'
2021-05-12 13:14:39,906 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-100025 as C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-100025.FAILED
Traceback (most recent call last):
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
[... cut ...]
2021-05-12 13:21:39,349 - mu.virtual_environment:669(upgrade_pip) INFO: Upgraded pip
2021-05-12 13:21:39,349 - mu.virtual_environment:711(install_baseline_packages) INFO: Installing baseline packages.
2021-05-12 13:21:39,350 - mu.virtual_environment:712(install_baseline_packages) INFO: C:\Users\alice\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-05-12 13:21:39,352 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
</code></pre></details>

carlosperate, 2021-05-12:
Looking at the list of wheels that it tries to install:
```
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, pywin32, parso, tornado, pyzmq, python-dateutil, Pygments, pycparser, prompt-toolkit, pickleshare, jupyter-core, jedi, decorator, colorama, backcall, MarkupSafe, jupyter-client, ipython, cffi, Werkzeug, reedsolo, QtPy, pyserial, pygame, numpy, Jinja2, itsdangerous, ipykernel, ecdsa, cryptography, click, bitstring, qtconsole, pgzero, Flask, esptool, adafruit-board-toolkit
```

And what it actually installed:

```
Package          Version
---------------- -------
backcall         0.2.0
colorama         0.4.4
decorator        5.0.7
ipython          7.22.0
ipython-genutils 0.2.0
jedi             0.18.0
jupyter-client   6.1.12
jupyter-core     4.7.1
MarkupSafe       1.1.1
parso            0.8.2
pickleshare      0.7.5
pip              21.1.1
prompt-toolkit   3.0.18
pycparser        2.20
Pygments         2.8.1
python-dateutil  2.8.1
pywin32          300
pyzmq            22.0.3
setuptools       56.0.0
six              1.15.0
tornado          6.1
traitlets        5.0.5
wcwidth          0.2.5
wheel            0.36.2
```

It looks like it failed at the `cffi` package.

@Alice-star-lgtm could you try running this command in a "Command Prompt" (you can open one by searching for `cmd.exe` in the Windows start menu) and copy/paste the output here?

```
C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439.FAILED\scripts\pip.exe install "c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl"
```

Alice-star-lgtm, 2021-05-12:
As Requested
C:\Users\alice>C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439.FAILED\scripts\pip.exe install "c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl"Fatal error in launcher: Unable to create process using '"c:\users\alice\appdata\local\python\mu\mu_venv-38-20210512-131439\scripts\python.exe"  "C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439.FAILED\scripts\pip.exe" install "c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl"': The system cannot find the file specified.

Alice-star-lgtm, 2021-05-12:
So, Mu appears to have installed despite error messages???!!!!

carlosperate, 2021-05-12:
Thank you @Alice-star-lgtm!

Unfortunately I didn't take in consideration that after the venv folder is renamed (by Mu when it encounters these errors) then things don't work any more inside it.

Could you try again but first renaming the folder:
1) Rename the `C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439.FAILED` to `C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439`
    - This is basically removing `.FAILED` from the `mu_venv-38-20210512-131439.FAILED` folder name.
    - I think this command should do it:
      ```
      ren "C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439.FAILED" "C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439"
      ```
2) Run:
    ```
    C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439\scripts\pip.exe install "c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl"
    ```

Does Mu run correctly already? Even if that was the casee it'd be great if you could run this test, to see if that step can still cause issues.

Alice-star-lgtm, 2021-05-12:
Hi
1. The first task returns the message "The syntax of the command is incorrect"2. Yes, Mu runs correctly.  If you can fix the above I'll happily run the test anyway

carlosperate, 2021-05-12:
Thanks for testing again @Alice-star-lgtm, sorry my command wasn't correct.

I'm not really running Windows at the moment, would you be able to navigate to the `C:\Users\alice\AppData\Local\python\mu\` folder and rename the folder `mu_venv-38-20210512-131439.FAILED` to `mu_venv-38-20210512-131439`? Just need to remove the `.FAILED` part.

After that the second command *should* work (well, it'll hopefully fail with an error message we can analyse):
```
C:\Users\alice\AppData\Local\python\mu\mu_venv-38-20210512-131439\scripts\pip.exe install "c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl"
```

carlosperate, 2021-05-18:
Just noticed the wheel installation stops at different points on different attempts:

```
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, pywin32, parso, tornado, setuptools, pyzmq, python-dateutil, Pygments, pycparser, prompt-toolkit, pickleshare, jupyter-core, jedi, decorator, colorama, backcall, MarkupSafe, jupyter-client, ipython, cffi, Werkzeug, reedsolo, QtPy, pyserial, pygame, numpy, Jinja2, itsdangerous, ipykernel, ecdsa, cryptography, click, bitstring, qtconsole, pgzero, Flask, esptool, adafruit-board-toolkit
```
```
Package          Version
---------------- -------
backcall         0.2.0
bitstring        3.1.7
cffi             1.14.5
click            7.1.2
colorama         0.4.4
cryptography     3.4.7
decorator        5.0.7
ecdsa            0.16.1
esptool          3.0
Flask            1.1.2
ipykernel        5.5.3
ipython          7.22.0
ipython-genutils 0.2.0
itsdangerous     1.1.0
jedi             0.18.0
Jinja2           2.11.3
jupyter-client   6.1.12
jupyter-core     4.7.1
MarkupSafe       1.1.1
numpy            1.20.2
parso            0.8.2
pgzero           1.2.1
pickleshare      0.7.5
pip              21.1.1
prompt-toolkit   3.0.18
pycparser        2.20
pygame           2.0.1
Pygments         2.8.1
pyserial         3.5
python-dateutil  2.8.1
pywin32          300
pyzmq            22.0.3
qtconsole        4.7.4
QtPy             1.9.0
reedsolo         1.5.4
setuptools       56.0.0
six              1.15.0
tornado          6.1
traitlets        5.0.5
wcwidth          0.2.5
Werkzeug         1.0.1
```

In this case it almost made it to the end, left out the last package `adafruit-board-toolkit`.

carlosperate, 2021-08-05:
Hi @Alice-star-lgtm, we've refactor how some of this venv creation works in beta.5, would you be able to install the latest release and try again?

carlosperate, 2022-05-31:
Closing due to lack of activity.