mu-editor/mu #1561: MU on Raspberrypi pico saving on every stroke.

fleek, 2021-05-12:
Currently MU writes to the raspberry pi, saving on every keystroke.

The raspberry pi pico and others have limited write cycles, same for even all media, even more for writable ROMs, why is MU writing every keystroke?

ntoll, 2021-05-13:
Hi @fleek can you attach the log file that shows Mu is saving on every keystroke..? (See: https://codewith.mu/en/howto/1.1/read_logs)

**At the request of the Raspberry Pi Foundation**, Mu will save a copy of the file every n seconds if the file has been changed. Perhaps that is what you're observing. I agree with you that this is not a sensible idea for a device with limited write cycles and I think I'll quietly switch this off. Just need to confirm via the logs that it's this "feature" that's causing the behaviour you're observing.

Thanks.

carlosperate, 2021-05-13:
Thanks for the report @fleek!

As far as I know the Pi Pico can be used in two modes:
- RP2040 mode: If the device is running MicroPython. In this case the files are saved on the PC and the programme is run on the device via REPL when the "Run" button is pressed.
- CircuitPython mode: If the device is running CircuitPython. The file is loaded from the CIRCUITPY MSD drive and should only be written to flash when the file is manually saved via "Save" button or `Ctrl+S`.

Which mode are you using? If it's saving to flash more often then this, then that is likely a bug in Mu we will need to fix. If you have the logs that would help us immensely.