mu-editor/mu #1564: Crash Report

dboe, 2021-05-13:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_CA
Platform: Linux 5.8.0-50-generic #56~20.04.1-Ubuntu SMP Mon Apr 12 21:46:35 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/dave/.local/lib/python3.8/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/dave/.local/lib/python3.8/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/dave/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/dave/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 627, in create
    self.install_jupyter_kernel()
  File "/home/dave/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 696, in install_jupyter_kernel
    raise VirtualEnvironmentCreateError("Unable to install kernel")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to install kernel

```

dboe, 2021-05-13:
ran from home dir solved it

closed 2021-05-13 by dboe

carlosperate, 2021-05-13:
Thanks for the report @dboe!
Could you let us know how did you run it originally and how did you run it when you fixed it? It might help anybody searching for a similar issue here.

dboe, 2021-05-14:
Seems to only run correctly from my home directory where libraries are installed. Probably a pathing issue

carlosperate, 2021-05-14:
Not sure I understand, did you use a virtualenvironemt, install from source code, or a "normal" pip install?

If you've installed it using `pip3 install mu-editor==1.1.0b4` then it would be installed in your "system" Python, so the location of where those commands are run should not have made a difference?
The reason I ask is because maybe we have overlooked something and if the location where the command is run affects Mu operation then it sounds like a bug we need to look into.

dboe, 2021-05-14:
I can't reproduce the problem, so you are probably right, but I am not sure at this point how we could get back to that situation.