mu-editor/mu #1565: Mu Editor Crash Report after first install

Inkad, 2021-05-14:
After I installed Mu Editor and started it, it crahed.
Uninstalled it, rebooted Windows pc, reinstalled it, but with the same crash report.

Does anyone know what went wrong?

As I want to start using Mu, I am rather new to programming :)

[mu.log.txt](https://github.com/mu-editor/mu/files/6480208/mu.log.txt)

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: nl_NL
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\van der Velde\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\van der Velde\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\van der Velde\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\van der Velde\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\van der Velde\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

tjguk, 2021-05-14:
Thanks for uploading the crash report with the logs. I'm sorry you're tripping up on our setup on your first attempt.

It's not clear to me from the logs what went wrong, but something goes wrong when the code tries to upgrade pip. It's a bit of a gamble, but would you mind trying with the previous beta:

https://github.com/mu-editor/mu/releases/download/1.1.0-beta.3/Mu-Editor-Win64-1.1.0b3.msi

which doesn't have the upgrade-pip functionality built in?

Inkad, 2021-05-15:
tjguk, thank you for your replay.

I uninstalled the initial version, downloaded the version you mentioned and installed this one.

Unfortunately, on the first start up there is an error also. See log file in the attachement.
[mu.lognew.txt](https://github.com/mu-editor/mu/files/6487750/mu.lognew.txt)

Do you know what went wrong this time?

carlosperate, 2021-08-06:
Hi @Inkad, the beta.5 has multiple fixes in this area, would you be able to install this version and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

DeCerveja, 2021-09-09:
Hi Carlos, I wanted to know if the assets provided are compatible with MacOS? I'm experiencing a similar problem as OP and would like to give the asset you posted a try!

carlosperate, 2021-09-21:
Hi @DeCerveja, if you are still experiencing an issue in macOS could you open a new GitHub issue detailing the problem?

There are a few different issues in this area that look very similar, so it'll be better to keep things separate.

@Inkad have you been able to test beta.5? There has been some fixes in this area that could have resolved the problem you were experiencing: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5