mu-editor/mu #1566: Install issue

Labuscj1, 2021-05-16:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Maryke\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Maryke\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Maryke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Maryke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Maryke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 344, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Maryke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 798, in setup
    os.makedirs(wd)
  File "C:\Users\Maryke\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Maryke\\mu_code'

```

carlosperate, 2021-06-04:
Hi  @Labuscj1, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\Maryke\AppData\Local\python\mu\logs`.

carlosperate, 2021-09-21:
I'll close this due to lack of activity, it looks like the race condition already tracked in #1755.

@Labuscj1 if you are still experiencing this problem please reopen this issue, thanks!

closed 2021-09-21 by carlosperate