mu-editor/mu #1569: Crash Report

gitness, 2021-05-16:
[mu.log.2021-05-16.pdf](https://github.com/mu-editor/mu/files/6489341/mu.log.2021-05-16.pdf)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: pt_PT
Platform: Linux 5.4.73v64 #2 SMP PREEMPT Tue Nov 3 16:11:05 CET 2020 aarch64

```
Traceback (most recent call last):
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/wheels/__init__.py", line 137, in download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1

```

carlosperate, 2021-05-17:
Thanks for the report and for attaching the log @gitness!

Would you be able to attach again the original log file? It's easier for us to read and parse the doc in plain text, as the PDF is really difficult to read and copy/pasting the contents has some issues with the line breaks.

carlosperate, 2021-05-18:
We'll continue this issue in https://github.com/mu-editor/mu/issues/1577.

closed 2021-05-18 by carlosperate

gitness, 2021-05-19:
Tanks for the answer. I have already resend others reports (including  
original logs) about the same subject. It evaluated after "sudo apt  
install libsdl2-dev" but was not solved. 

 Best regards,

Santiago.

----- Mensagem de Carlos Pereira Atencio ***@***.***> ---------

  Data: Tue, 18 May 2021 14:17:20 -0700

----- Fim da mensagem de Carlos Pereira Atencio  

Ligações:
---------
[3]