mu-editor/mu #1570: Mu installed but window not appearing when run

BarryButler, 2021-05-17:
Mu suddenly has stopped working when the icon is clicked (mu-editor_1.1.0-alpha.2_win64).  I removed the program and reinstalled it and have the same problem.

When I click on the Mu icon, the splash screen appears and the icon is put on my task bar but the Mu window does not appear.  Passing the mouse over the task bar icon shows that the program is loaded.

I cannot get into the window, so cannot get to the log file.  Let me know where it is located

BarryButler, 2021-05-17:
Found the mu.log

[mu.log](https://github.com/mu-editor/mu/files/6490890/mu.log)

carlosperate, 2021-08-10:
Thanks for the report and the log @BarryButler!

Looking at the log it seems like sometimes the window does open and you are able to close it via the "Quit" button is that right?

Alpha.2 is a very old release, if could you also try the latest beta and see if you still have this issue? https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

BarryButler, 2021-09-25:
> @BarryButler<https://github.com/BarryButler> have you been able to test the latest beta.5 to see if it works on your computer? the alpha.2 is very old and hopefully this issue is already fixed.
>
>
> ***************************************************************************************************
>
> IMPORTANT: This email and any attachments may contain legally privileged, confidential or private information, and may be protected by copyright. You may only use or disclose this information if you are the intended recipient(s) and if you use it in an authorised way. No other person is allowed to use, review, alter, transmit, disclose, distribute, print or copy this email and any attachments without appropriate authorisation.
>
> If you are not the intended recipient(s) and the email was sent to you by mistake, please notify the sender immediately by return email or phone, destroy any hardcopies of this email and any attachments and delete it from your system. Any legal privilege and confidentiality attached to this email is not waived or destroyed by that mistake.
>
> The Department of Education carries out monitoring, scanning and blocking of emails and attachments sent from or to addresses within the Department of Education for the purposes of operating, protecting, maintaining and ensuring appropriate use of its computer network. It is your responsibility to ensure that this email does not contain and is not affected by computer viruses, defects or interference by third parties or replication problems (including incompatibility with your computer system).
>
> The Department of Education does not accept any responsibility for any loss or damage that may result from reliance on, or the use of, any information contained in the email and any attachments.
>
> ***************************************************************************************************

Everything works great.

Regards,

Barry Butler
Teacher (Digital Technology, Aviation, Maths, Science)

[2019 Email Signature Block]
------------------------

This message (including attachments) is intended for the addressee named above. It may also be confidential, privileged and/or subject to copyright. If you wish to forward this message to others, you must first obtain the permission of the author. If you are not the addressee named above, you must not disseminate, copy, communicate or otherwise use or take any action in reliance on this message. You understand that any privilege or confidentiality attached to this message is not waived, lost or destroyed because you have received this message in error. If you have received this message in error please notify the sender and delete from any computer. Unless explicitly attributed, the opinions expressed in this message do not necessarily represent the official position or opinions of the State of Queensland or the Department of Education, Training and Employment. Whilst all care has been taken, the Department of Education, Training and Employment disclaims all liability for loss or damage to person or property arising from this message being infected by computer virus or other contamination.

carlosperate, 2021-09-26:
That's great, thanks for the update @BarryButler!

closed 2021-09-26 by carlosperate