mu-editor/mu #1571: Mu Editor 1.1.0b4 does not add the file extension .py when saving a new file for the first time on Linux

amos1969, 2021-05-17:
**Steps to recreate:**

1. Install the latest beta via pip in a virtual environment (Kubuntu 21.04 using a Python 3.8.10 virtual environment via pyenv).
2. Start the Mu Editor as normal.
3. Click on New to create a new tab.
4. Click on Save and in the resulting dialog box, enter a file name for the file, but do not add the .py extension and hit enter.

**Expected behaviour**

File saves with the .py extension added automatically.

**Actual behaviour**

File saves as a plain text file with no file extension.

This behaviour is different to that experienced on the Windows version which does add the file extension by default.

Tahnks

ntoll, 2021-05-17:
Aha... nice spot, I've just recreated the problem. Lemme work out a fix.

ntoll, 2021-05-17:
So, it appears that the results of calling `QFileDialog.getSaveFileName` are different between OSs. On Windows and Mac the system default file handler is used, and presumably works as expected given the following call (note the default file type filters). On Linux some other mechanism is used for for entering a new filename. I suspect, because of the inclusion of `Other (*.*)` in the filter, you can now save files without the expected `.py` extension (this was introduced so web mode could open CSS and HTML files).

```
path, _ = QFileDialog.getSaveFileName(
            self.widget,
            "Save file",
            folder if self.previous_folder is None else self.previous_folder,
            "Python (*.py);;Other (*.*)",
            "Python (*.py)",
        )
```

I'll add a fix into the view method that calls the file dialog, to ensure a `.py` is added to the end of a file that doesn't already have a file extension.

ntoll, 2021-05-17:
@amos1969 the fix is in [this PR](https://github.com/mu-editor/mu/pull/1572) and should be in the next release once it's reviewed and checked.

amos1969, 2021-05-17:
Just checked it out and it worked exactly as expected, for me.

Thanks
Dave

PR #1572 (2021-06-19): Ensure, if no file extension provided, a .py one is used. 
Fixes #1571.
commit 56838980 (2021-06-19): Ensure, if no file extension provided, a .py one is used. Fixes #1571.
closed 2021-06-19 by ntoll