mu-editor/mu #1574: Crash Report

lpoolehfed, 2021-05-17:
[mu.log](https://github.com/mu-editor/mu/files/6496560/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-05-18:
@lpoolehfed thanks for the report and specially for attaching the log!

Looking at the wheels Mu packs and try to install:
```
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, pywin32, parso, tornado, setuptools, pyzmq, python-dateutil, Pygments, pycparser, prompt-toolkit, pickleshare, jupyter-core, jedi, decorator, colorama, backcall, MarkupSafe, jupyter-client, ipython, cffi, Werkzeug, reedsolo, QtPy, pyserial, pygame, numpy, Jinja2, itsdangerous, ipykernel, ecdsa, cryptography, click, bitstring, qtconsole, pgzero, Flask, esptool, adafruit-board-toolkit
```

And the installed packages:

```
Package          Version
---------------- -------
ipython-genutils 0.2.0
jedi             0.18.0
jupyter-core     4.7.1
parso            0.8.2
pickleshare      0.7.5
pip              21.1.1
prompt-toolkit   3.0.18
pycparser        2.20
Pygments         2.8.1
python-dateutil  2.8.1
pywin32          300
pyzmq            22.0.3
setuptools       56.0.0
six              1.15.0
tornado          6.1
traitlets        5.0.5
wcwidth          0.2.5
```

It looks like it stopped at `decorator`. However different attempts fail at different point:

```
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, pywin32, parso, tornado, setuptools, pyzmq, python-dateutil, Pygments, pycparser, prompt-toolkit, pickleshare, jupyter-core, jedi, decorator, colorama, backcall, MarkupSafe, jupyter-client, ipython, cffi, Werkzeug, reedsolo, QtPy, pyserial, pygame, numpy, Jinja2, itsdangerous, ipykernel, ecdsa, cryptography, click, bitstring, qtconsole, pgzero, Flask, esptool, adafruit-board-toolkit
```
```
Package          Version
---------------- -------
backcall         0.2.0
cffi             1.14.5
colorama         0.4.4
decorator        5.0.7
ipython          7.22.0
ipython-genutils 0.2.0
jedi             0.18.0
jupyter-client   6.1.12
jupyter-core     4.7.1
MarkupSafe       1.1.1
parso            0.8.2
pickleshare      0.7.5
pip              21.1.1
prompt-toolkit   3.0.18
pycparser        2.20
Pygments         2.8.1
pyserial         3.5
python-dateutil  2.8.1
pywin32          300
pyzmq            22.0.3
QtPy             1.9.0
reedsolo         1.5.4
setuptools       56.0.0
six              1.15.0
tornado          6.1
traitlets        5.0.5
wcwidth          0.2.5
Werkzeug         1.0.1
wheel            0.36.2
```

This crash report had the same issue: https://github.com/mu-editor/mu/issues/1559

@tjguk can we make sure each wheel is installed individually (it'll need the `--no-deps` flag).

carlosperate, 2021-05-27:
Thank you very much for the report @lpoolehfed!

I think we might have several crash reports with this similar issue, so I'll merge all of them into a single issue: https://github.com/mu-editor/mu/issues/1600#issuecomment-849815968

We can continue the conversation over there, so to keep things tidy I'll close this one.

Thanks again!

closed 2021-05-27 by carlosperate