mu-editor/mu #1576: Cannot install pygame in Raspbian Stretch

carlosperate, 2021-05-18:
Unfortunately there are no Arm Stretch wheels for pygame v2.0.* in piwheels , which is the version required by the latest pgzero.

We could add a workaround in `mu.wheels` to install `pygame==2.0.0.dev14` (last available wheel, but it's a "pre-release") in this platform before it attempts to install `pgzero`, although ideally we should try to get the Arm wheels in piwheels as that might benefit everybody.

carlosperate, 2021-05-18:
Ticket in piwheels repo: https://github.com/piwheels/packages/issues/208

ntoll, 2021-05-18:
Help us @bennuttall, you're our only hope. :-P

carlosperate, 2021-05-18:
Turns out we also need to be careful about numpy, as a normal `pip install` tries to build from source as well: https://www.piwheels.org/project/numpy/
However, since pgzero can use any version the `--prefer-binary` is enough (`--only-binary ":all"` also works).

I was hoping adding the `--pre` flag would be enough to use pygame `2.0.0.dev14`, however it looks like when it compares  versions we `2.0.0.dev14 < 2.0.0`, so we would need to manually install the dependencies (https://github.com/lordmauve/pgzero/blob/1.2.1/setup.py#L9-L12) and then install pgzero with the `--no-deps` flag:
```
pip install "pygame==2.0.0.dev14"
pip install numpy --prefer-binary
pip install "pgzero==1.2.1" --no-deps
```

bennuttall, 2021-05-18:
> Help us @bennuttall, you're our only hope. :-P

Ha, you must be in trouble

I've posted the traceback on the piwheels issue (thanks for that - much appreciated) and pinged the maintainer.

ntoll, 2021-05-19:
Thank you @bennuttall - you're a :star:

carlosperate, 2021-05-28:
The Arm pygame wheels for Stretch are now in piwheels 🎉
Thanks @bennuttall, @illume, @robertpfeiffer, @ankith26, and everybody else that contributed!

As far as numpy is concerned, looks like Mu is downloading the latest wheel available (1.18.5) even without the specific flags, so in the end it is not trying to build from source the latest version (1.20.3).

closed 2021-05-28 by carlosperate

bennuttall, 2021-05-28:
> As far as numpy is concerned, looks like Mu is downloading the latest wheel available (1.18.5) even without the specific flags, so in the end it is not trying to build from source the latest version (1.20.3).

That'll be the `data-requires-python` attribute on the simple index (view source on [/simple/numpy](https://www.piwheels.org/simple/numpy/) specifying the required python version.

carlosperate, 2021-05-28:
Ahh, you are 100% right, v1.19.0 dropped 3.5 ([`Requires: Python >=3.6`](https://pypi.org/project/numpy/1.19.0/)), so the latest compatible version is 1.18.5 and that's the wheel it fetches from piwheels.

bennuttall, 2021-05-28:
Ah but does the version of pip in Stretch (from apt, not upgraded) support requires-python? Might be worth checking.

carlosperate, 2021-05-28:
As this is inside a venv created with the `virtualenv` module, then pip gets automatically updated to the latest version available online or the one packed inside the `virtualenv` module.
This also means it gets the right pip version for Python 3.5, since the latest releases are Py3.6+ and pip v9 doesn't implement `requires-python` and so it breaks itself fetching the latest.