mu-editor/mu #1577: Crash Report

gitness, 2021-05-18:
[mu.log](https://github.com/mu-editor/mu/files/6504240/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: pt_PT
Platform: Linux 5.4.73v64 #2 SMP PREEMPT Tue Nov 3 16:11:05 CET 2020 aarch64

```
Traceback (most recent call last):
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/wheels/__init__.py", line 137, in download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/home/santiago/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1

```
  I am very sorry to insist. I am a newbie in this matter  and need help.  I  kept going to try to get mu-editor to work and, after fiding "sudo apt install libsdl2-dev" in de net the error to get pgzero 1.2.1 installed seemed to change. I think the env configuration is not well completed. Please help. Many tanks.
[mu.log](https://github.com/mu-editor/mu/files/6504300/mu.log)

carlosperate, 2021-05-18:
Hi @gitness,

Could you let us know a bit more about your environment? Is this running in a Single Board Computer? If so which one?
What Operating System and version are you running?
What commands did you use exactly to install Mu? did you have to use `sudo`?

carlosperate, 2021-05-28:
Hi @gitness,

Just wondering if you were able to resolve this issue?
Are you using an SBC like the Raspberry Pi? In that case what OS are you using, Raspberry Pi OS?

gitness, 2021-05-28:
Hi,

      Many tanks for answering. I tried two OS: RaspberryOS/Buster and  
Ubuntu 2.10 for RPI. At first try, both OS have shown the same problem  
at mu-editor startup time: pgzero 1.2.1 or superior couldn't be  
installed! After trying some hypotetical solution from Internet,  
things gradually changed, and finally stuck. Now, I purged mu-editor.  
In the midle of the all thing, I tried reinstall the mu versions that  
I could used before try to upgrade it for the use of circuitpython  
sake; but those versions stoped to run too. Luckly, I can work with  
micropython using Thonny; generic circuitpython interpreter is  
installed too, but it can't find ttyACM0 port. 

     Sorry for my too long message and cripled english.

     Best regards,

JS.

----- Mensagem de Carlos Pereira Atencio ***@***.***> ---------

  Data: Fri, 28 May 2021 07:08:56 -0700

----- Fim da mensagem de Carlos Pereira Atencio  

Ligações:
---------
[3]

gitness, 2021-05-28:
Hi,

      Many tanks for answering. I tried two OS: RaspberryOS/Buster and  
Ubuntu 20.10 for RPI. At first try, both OS have shown the same  
problem at mu-editor startup time: pgzero 1.2.1 or superior couldn't  
be installed! After trying some hypotetical solution from Internet,  
things gradually changed, and finally stuck. Now, I purged mu-editor.  
In the midle of the all thing, I tried reinstall the mu versions that  
I could used before try to upgrade it for the use of circuitpython  
sake; but those versions stoped to run too. Luckly, I can work with  
micropython using Thonny; generic circuitpython interpreter is  
installed too, but it can't find ttyACM0 port. 

     Sorry for my too long message and cripled english.

     Best regards,

JS.

----- Mensagem de Carlos Pereira Atencio ***@***.***> ---------

  Data: Fri, 28 May 2021 07:08:56 -0700

carlosperate, 2021-08-05:
Hi @gitness,

There is a new beta.5 release that has reworked how some of this works. Would you be able to instal the latest beta and try again? hopefully that should resolve your issue.

carlosperate, 2021-09-02:
Hi @gitness, I was wondering if you had a chance to try beta.5 and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5
`pip install mu-editor==1.1.0b5`

Are you using an Single Board Computer like the Raspberry Pi? In that case what OS are you using, Raspberry Pi OS, Ubuntu or something else?

carlosperate, 2021-12-17:
Closing due to lack of activity.
If you are still experiencing this problem please reopen this issue and we can continue looking into it.
And thank you again for the report!

closed 2021-12-17 by carlosperate