mu-editor/mu #1586: Crash Report

gbrault, 2021-05-22:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 754, in on_ls_fail
    self.disable()
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 693, in disable
    self.microbit_fs.setDisabled(True)
RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted

```

carlosperate, 2021-05-24:
Thanks for the report @gbrault!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

gbrault, 2021-05-24:
Hello,
Find attached all the logs I have from Mu.
Best regards
[mu.log](https://github.com/mu-editor/mu/files/6531429/mu.log)
[mu.log.2021-05-22.log](https://github.com/mu-editor/mu/files/6531430/mu.log.2021-05-22.log)

carlosperate, 2021-05-24:
Thanks @gbrault!

This is the interesting part from the logs:

```
2021-05-22 16:25:17,996 - mu.modes.esp:214(toggle_files) INFO: Toggle filesystem on.
2021-05-22 16:25:23,434 - mu.modes.base:686(ls) ERROR: Could not enter raw REPL.
Traceback (most recent call last):
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 683, in ls
    result = tuple(microfs.ls(self.serial))
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 185, in ls
    out, err = execute(["import os", "print(os.listdir())"], serial)
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 139, in execute
    raw_on(serial)
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 100, in raw_on
    flush_to_msg(serial, raw_repl_msg)
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 72, in flush_to_msg
    raise IOError("Could not enter raw REPL.")
OSError: Could not enter raw REPL.
2021-05-22 16:25:23,439 - mu.interface.main:942(show_message) DEBUG: There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your device and/or restarting Mu.
2021-05-22 16:25:23,439 - mu.interface.main:943(show_message) DEBUG: None
2021-05-22 16:25:46,506 - mu.interface.main:942(show_message) DEBUG: Could not find an attached ESP8266/ESP32
2021-05-22 16:25:46,506 - mu.interface.main:943(show_message) DEBUG: Please make sure the device is plugged into this computer.

The device must have MicroPython flashed onto it before the file system will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2021-05-22 16:26:11,511 - mu.logic:720(check_usb) INFO: esp device disconnected on port: COM4(VID: 0x10C4, PID: 0xEA60, manufacturer Silicon Laboratories)
2021-05-22 16:26:11,693 - mu.logic:737(check_usb) INFO: esp device connected on port: COM4(VID: 0x10C4, PID: 0xEA60, manufacturer: 'Silicon Laboratories')
2021-05-22 16:26:13,289 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 754, in on_ls_fail
    self.disable()
  File "C:\Users\gilbe\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 693, in disable
    self.microbit_fs.setDisabled(True)
RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted
```

Looks like it first tried to open the ESP model files panel, it could not entre the raw-REPL mode and asked the user to unplug and replug the device. Then after doing that Mu crashes with the `wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted` error.

The log was definitely very useful @gbrault! Hopefully we can replicate this problem following similar steps.

carlosperate, 2021-05-24:
I believe this is the same issue from https://github.com/mu-editor/mu/issues/1428, so to try to keep things tidy I will merge this report there to track the progress in a single place: https://github.com/mu-editor/mu/issues/1428#issuecomment-846946715

We can continue the debugging over there, thanks again for taking the time to report this and provide the additional information @gbrault!

closed 2021-05-24 by carlosperate