mu-editor/mu #1588: Crash Report

DanilovEA, 2021-05-23:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: ru_RU
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 711, in check_usb
    detected = mode.find_devices(with_logging=False)
  File "C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 423, in find_devices
    device = self.compatible_board(port)
  File "C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\circuitpython.py", line 286, in compatible_board
    for comport in circuitpython_serial.repl_comports():
  File "C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 61, in repl_comports
    port for port in comports() if port.interface.startswith("CircuitPython CDC ")
  File "C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 46, in comports
    ports = _list_ports_windows.comports()
  File "C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 458, in comports
    return list(iterate_comports())
  File "C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 342, in iterate_comports
    info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'

```

carlosperate, 2021-05-24:
Thanks for the report @DanilovEA!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

DanilovEA, 2021-05-24:
[mu.log](https://github.com/mu-editor/mu/files/6532101/mu.log)

carlosperate, 2021-05-24:
Thank you for the log @DanilovEA!

Looks like you might have an older version pyserial installed somehow. How did you install the Mu editor? With the installer or pip?

DanilovEA, 2021-05-24:
I'm install mu-editor by installer, because Pip can't find version 1.1.0-beta-4. I have last version of pyserial==3.5. I want try work with Pico.

carlosperate, 2021-05-24:
Thank you @DanilovEA!
Could you run this in a command prompt (you can type `cmd` in the Windows start menu to find it) and copy/paste the result here?

```
"C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list
```

DanilovEA, 2021-05-24:
Well done. On this path pyserial was 3.4. I upgrade it. And mu-editor work well.

carlosperate, 2021-05-24:
Awesome! I'm glad that worked.

However, I'm not sure how pyserial 3.4 made it into the `C:\Users\flash\AppData\Local\Programs\Mu Editor\Python\site_packages` installation, as the installer should have packed 3.5?
Do you still have the installer you used to install this version of Mu?

DanilovEA, 2021-05-24:
[image]
Hm, my installer include pyserial-3.5. Later I use old version mu-editor.

carlosperate, 2021-05-26:
Ah, okay, so an older version was added accidentally by using an older mu-editor. In that case I think we can close this issue.
Thanks for looking into this and all the testing!

closed 2021-05-26 by carlosperate