mu-editor/mu #1589: Crash Report

w4jkl, 2021-05-23:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64

```
Traceback (most recent call last):
  File "/home/john/.local/lib/python3.7/site-packages/mu/logic.py", line 698, in check_usb
    def check_usb(self):
KeyboardInterrupt

```
Locked Up

w4jkl, 2021-05-23:
Forgot to add this from the command line:

john@debian:~$ mu-editor
2021-05-23 16:25:21,704 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-23 16:25:21,704 - root:266(run) INFO: uname_result(system='Linux', node='debian', release='4.19.0-16-amd64', version='#1 SMP Debian 4.19.181-1 (2021-03-19)', machine='x86_64', processor='')
2021-05-23 16:25:21,705 - root:267(run) INFO: Python path: ['/home/john/.local/bin', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/home/john/.local/lib/python3.7/site-packages', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2021-05-23 16:25:21,705 - root:268(run) INFO: Language code: en_US
2021-05-23 16:25:21,974 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-23 16:25:22,020 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-23 16:25:22,020 - mu.virtual_environment:542(ensure_path) INFO: Virtual Environment found at: /home/john/.local/share/mu/mu_venv-37-20210523-155906
2021-05-23 16:25:22,029 - mu.virtual_environment:555(ensure_interpreter) INFO: Interpreter found at: /home/john/.local/share/mu/mu_venv-37-20210523-155906/bin/python
2021-05-23 16:25:22,104 - mu.virtual_environment:382(run_subprocess) DEBUG: Process returned 0; output: 37
2021-05-23 16:25:22,105 - mu.virtual_environment:585(ensure_interpreter_version) INFO: Both interpreters at version 37
2021-05-23 16:25:22,110 - mu.virtual_environment:611(ensure_pip) INFO: Pip found at: /home/john/.local/share/mu/mu_venv-37-20210523-155906/bin/pip
2021-05-23 16:25:22,111 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-05-23 16:25:22,183 - mu.virtual_environment:382(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-23 16:25:22,184 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: flask
2021-05-23 16:25:22,694 - mu.virtual_environment:382(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-23 16:25:22,695 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-05-23 16:25:22,765 - mu.virtual_environment:382(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-23 16:25:22,765 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-05-23 16:25:22,898 - mu.virtual_environment:382(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-23 16:25:22,898 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: adafruit_board_toolkit
2021-05-23 16:25:22,964 - mu.virtual_environment:382(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-23 16:25:22,965 - mu.virtual_environment:496(ensure_and_create) INFO: Valid virtual environment found at /home/john/.local/share/mu/mu_venv-37-20210523-155906
2021-05-23 16:25:22,965 - mu.settings:169(save) DEBUG: Saving to /home/john/.local/share/mu/venv.json
2021-05-23 16:25:22,971 - mu.virtual_environment:511(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x7fa600d1eaf8>; Splash Handler; <SplashLogHandler (INFO)>
2021-05-23 16:25:22,988 - mu.logic:758(__init__) INFO: Setting up editor.
2021-05-23 16:25:22,991 - mu.logic:778(__init__) INFO: Log directory: /home/john/.cache/mu/log
2021-05-23 16:25:22,992 - mu.logic:779(__init__) INFO: Data directory: /home/john/.local/share/mu
2021-05-23 16:25:22,992 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-05-23 16:25:22,993 - mu.modes.base:63(get_default_workspace) INFO: Using workspace /home/john/mu_code from settings file
2021-05-23 16:25:23,095 - mu.modes.base:63(get_default_workspace) INFO: Using workspace /home/john/mu_code from settings file
2021-05-23 16:25:23,096 - mu.logic:1521(change_mode) INFO: Workspace directory: /home/john/mu_code
2021-05-23 16:25:23,096 - mu.logic:867(restore_session) DEBUG: <SessionSettings from /home/john/.local/share/mu/session.json>
2021-05-23 16:25:23,110 - mu.logic:1469(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'pyboard', 'debugger', 'pygamezero', 'lego', 'pico']

w4jkl, 2021-05-23:
I tried a re-install:

john@debian:~$ pip3 install mu-editor==1.1.0b4
Requirement already satisfied: mu-editor==1.1.0b4 in ./.local/lib/python3.7/site-packages (1.1.0b4)
Requirement already satisfied: QScintilla==2.11.3; "arm" not in platform_machine and "aarch" not in platform_machine in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (2.11.3)
Requirement already satisfied: PyQt5==5.13.2; "arm" not in platform_machine and "aarch" not in platform_machine in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (5.13.2)
Requirement already satisfied: appdirs>=1.4.3 in /usr/lib/python3/dist-packages (from mu-editor==1.1.0b4) (1.4.3)
Requirement already satisfied: wheel in /usr/lib/python3/dist-packages (from mu-editor==1.1.0b4) (0.32.3)
Requirement already satisfied: qtconsole==4.7.7 in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (4.7.7)
Requirement already satisfied: black>=19.10b0; python_version > "3.5" in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (21.5b1)
Requirement already satisfied: semver>=2.8.0 in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (2.13.0)
Requirement already satisfied: flake8>=3.8.3 in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (3.9.2)
Requirement already satisfied: adafruit-board-toolkit>=1.0.1 in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (1.1.0)
Requirement already satisfied: virtualenv>=16.0.0 in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (20.4.6)
Requirement already satisfied: PyQtChart==5.13.1; "arm" not in platform_machine and "aarch" not in platform_machine in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (5.13.1)
Requirement already satisfied: jupyter-client<6.2,>=4.1 in /usr/lib/python3/dist-packages (from mu-editor==1.1.0b4) (5.2.3)
Requirement already satisfied: pyserial>=3.5 in ./.local/lib/python3.7/site-packages (from mu-editor==1.1.0b4) (3.5)
Requirement already satisfied: nudatus>=0.0.3 in /usr/lib/python3/dist-packages (from mu-editor==1.1.0b4) (0.0.4)
Requirement already satisfied: PyQt5_sip<13,>=4.19.19 in ./.local/lib/python3.7/site-packages (from PyQt5==5.13.2; "arm" not in platform_machine and "aarch" not in platform_machine->mu-editor==1.1.0b4) (12.9.0)
Requirement already satisfied: traitlets in /usr/lib/python3/dist-packages (from qtconsole==4.7.7->mu-editor==1.1.0b4) (4.3.2)
Requirement already satisfied: ipython-genutils in /usr/lib/python3/dist-packages (from qtconsole==4.7.7->mu-editor==1.1.0b4) (0.2.0)
Requirement already satisfied: pygments in /usr/lib/python3/dist-packages (from qtconsole==4.7.7->mu-editor==1.1.0b4) (2.3.1)
Requirement already satisfied: ipykernel>=4.1 in /usr/lib/python3/dist-packages (from qtconsole==4.7.7->mu-editor==1.1.0b4) (4.9.0)
Requirement already satisfied: pyzmq>=17.1 in /usr/lib/python3/dist-packages (from qtconsole==4.7.7->mu-editor==1.1.0b4) (17.1.2)
Requirement already satisfied: jupyter-core in /usr/lib/python3/dist-packages (from qtconsole==4.7.7->mu-editor==1.1.0b4) (4.4.0)
Requirement already satisfied: qtpy in ./.local/lib/python3.7/site-packages (from qtconsole==4.7.7->mu-editor==1.1.0b4) (1.9.0)
Requirement already satisfied: click>=7.1.2 in ./.local/lib/python3.7/site-packages (from black>=19.10b0; python_version > "3.5"->mu-editor==1.1.0b4) (8.0.1)
Requirement already satisfied: regex>=2020.1.8 in ./.local/lib/python3.7/site-packages (from black>=19.10b0; python_version > "3.5"->mu-editor==1.1.0b4) (2021.4.4)
Requirement already satisfied: typing-extensions>=3.7.4; python_version < "3.8" in ./.local/lib/python3.7/site-packages (from black>=19.10b0; python_version > "3.5"->mu-editor==1.1.0b4) (3.10.0.0)
Requirement already satisfied: pathspec<1,>=0.8.1 in ./.local/lib/python3.7/site-packages (from black>=19.10b0; python_version > "3.5"->mu-editor==1.1.0b4) (0.8.1)
Requirement already satisfied: toml>=0.10.1 in ./.local/lib/python3.7/site-packages (from black>=19.10b0; python_version > "3.5"->mu-editor==1.1.0b4) (0.10.2)
Requirement already satisfied: mypy-extensions>=0.4.3 in ./.local/lib/python3.7/site-packages (from black>=19.10b0; python_version > "3.5"->mu-editor==1.1.0b4) (0.4.3)
Requirement already satisfied: typed-ast>=1.4.2; python_version < "3.8" in ./.local/lib/python3.7/site-packages (from black>=19.10b0; python_version > "3.5"->mu-editor==1.1.0b4) (1.4.3)
Requirement already satisfied: importlib-metadata; python_version < "3.8" in ./.local/lib/python3.7/site-packages (from flake8>=3.8.3->mu-editor==1.1.0b4) (4.0.1)
Requirement already satisfied: pycodestyle<2.8.0,>=2.7.0 in ./.local/lib/python3.7/site-packages (from flake8>=3.8.3->mu-editor==1.1.0b4) (2.7.0)
Requirement already satisfied: pyflakes<2.4.0,>=2.3.0 in ./.local/lib/python3.7/site-packages (from flake8>=3.8.3->mu-editor==1.1.0b4) (2.3.1)
Requirement already satisfied: mccabe<0.7.0,>=0.6.0 in ./.local/lib/python3.7/site-packages (from flake8>=3.8.3->mu-editor==1.1.0b4) (0.6.1)
Requirement already satisfied: distlib<1,>=0.3.1 in ./.local/lib/python3.7/site-packages (from virtualenv>=16.0.0->mu-editor==1.1.0b4) (0.3.1)
Requirement already satisfied: six<2,>=1.9.0 in /usr/lib/python3/dist-packages (from virtualenv>=16.0.0->mu-editor==1.1.0b4) (1.12.0)
Requirement already satisfied: filelock<4,>=3.0.0 in ./.local/lib/python3.7/site-packages (from virtualenv>=16.0.0->mu-editor==1.1.0b4) (3.0.12)
Requirement already satisfied: zipp>=0.5 in ./.local/lib/python3.7/site-packages (from importlib-metadata; python_version < "3.8"->flake8>=3.8.3->mu-editor==1.1.0b4) (3.4.1)

w4jkl, 2021-05-23:
where is the LOG FILE?

John

w4jkl, 2021-05-23:
found the log after reading more carefully...

2021-05-23 15:45:41,186 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2021-05-23 15:45:41,186 - root:113(run) INFO: uname_result(system='Linux', node='debian', release='4.19.0-16-amd64', version='#1 SMP Debian 4.19.181-1 (2021-03-19)', machine='x86_64', processor='')
2021-05-23 15:45:41,186 - root:114(run) INFO: Python path: ['/usr/share/mu-editor', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2021-05-23 15:45:41,186 - root:115(run) INFO: Language code: en_US
2021-05-23 15:45:41,452 - mu.logic:543(__init__) INFO: Setting up editor.
2021-05-23 15:45:41,454 - mu.logic:560(__init__) DEBUG: Creating directory: /home/john/.local/share/mu
2021-05-23 15:45:41,455 - mu.logic:316(get_admin_file_path) DEBUG: Creating admin file: /home/john/.local/share/mu/settings.json
2021-05-23 15:45:41,455 - mu.logic:562(__init__) INFO: Settings path: /home/john/.local/share/mu/settings.json
2021-05-23 15:45:41,456 - mu.logic:316(get_admin_file_path) DEBUG: Creating admin file: /home/john/.local/share/mu/session.json
2021-05-23 15:45:41,456 - mu.logic:563(__init__) INFO: Session path: /home/john/.local/share/mu/session.json
2021-05-23 15:45:41,456 - mu.logic:564(__init__) INFO: Log directory: /home/john/.cache/mu/log
2021-05-23 15:45:41,456 - mu.logic:565(__init__) INFO: Data directory: /home/john/.local/share/mu
2021-05-23 15:45:41,486 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2021-05-23 15:45:41,486 - mu.logic:582(setup) DEBUG: Creating directory: /home/john/mu_code
2021-05-23 15:45:41,487 - mu.logic:590(setup) DEBUG: Creating directory: /home/john/mu_code/images
2021-05-23 15:45:41,489 - mu.logic:597(setup) DEBUG: Creating directory: /home/john/mu_code/fonts
2021-05-23 15:45:41,489 - mu.logic:600(setup) DEBUG: Creating directory: /home/john/mu_code/sounds
2021-05-23 15:45:41,490 - mu.logic:603(setup) DEBUG: Creating directory: /home/john/mu_code/music
2021-05-23 15:45:41,693 - mu.logic:1148(change_mode) INFO: Workspace directory: /home/john/mu_code
2021-05-23 15:45:41,693 - mu.logic:625(restore_session) INFO: Restoring session from: /home/john/.local/share/mu/session.json
2021-05-23 15:45:41,694 - mu.logic:626(restore_session) DEBUG: {}
2021-05-23 15:45:41,694 - mu.logic:1105(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2021-05-23 15:48:31,969 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2021-05-23 15:48:31,970 - root:113(run) INFO: uname_result(system='Linux', node='debian', release='4.19.0-16-amd64', version='#1 SMP Debian 4.19.181-1 (2021-03-19)', machine='x86_64', processor='')
2021-05-23 15:48:31,970 - root:114(run) INFO: Python path: ['/usr/share/mu-editor', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2021-05-23 15:48:31,970 - root:115(run) INFO: Language code: en_US
2021-05-23 15:48:32,147 - mu.logic:543(__init__) INFO: Setting up editor.
2021-05-23 15:48:32,147 - mu.logic:562(__init__) INFO: Settings path: /home/john/.local/share/mu/settings.json
2021-05-23 15:48:32,148 - mu.logic:563(__init__) INFO: Session path: /home/john/.local/share/mu/session.json
2021-05-23 15:48:32,148 - mu.logic:564(__init__) INFO: Log directory: /home/john/.cache/mu/log
2021-05-23 15:48:32,148 - mu.logic:565(__init__) INFO: Data directory: /home/john/.local/share/mu
2021-05-23 15:48:32,175 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2021-05-23 15:48:32,295 - mu.logic:1148(change_mode) INFO: Workspace directory: /home/john/mu_code
2021-05-23 15:48:32,295 - mu.logic:625(restore_session) INFO: Restoring session from: /home/john/.local/share/mu/session.json
2021-05-23 15:48:32,295 - mu.logic:626(restore_session) DEBUG: {}
2021-05-23 15:48:32,295 - mu.logic:1105(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2021-05-23 15:53:06,121 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2021-05-23 15:53:06,121 - root:113(run) INFO: uname_result(system='Linux', node='debian', release='4.19.0-16-amd64', version='#1 SMP Debian 4.19.181-1 (2021-03-19)', machine='x86_64', processor='')
2021-05-23 15:53:06,121 - root:114(run) INFO: Python path: ['/usr/share/mu-editor', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2021-05-23 15:53:06,121 - root:115(run) INFO: Language code: en_US
2021-05-23 15:53:06,301 - mu.logic:543(__init__) INFO: Setting up editor.
2021-05-23 15:53:06,302 - mu.logic:562(__init__) INFO: Settings path: /home/john/.local/share/mu/settings.json
2021-05-23 15:53:06,302 - mu.logic:563(__init__) INFO: Session path: /home/john/.local/share/mu/session.json
2021-05-23 15:53:06,302 - mu.logic:564(__init__) INFO: Log directory: /home/john/.cache/mu/log
2021-05-23 15:53:06,302 - mu.logic:565(__init__) INFO: Data directory: /home/john/.local/share/mu
[... cut ...]
2021-05-23 19:14:16,213 - mu.logic:1469(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'pyboard', 'debugger', 'pygamezero', 'lego', 'pico']
2021-05-23 19:14:28,090 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/home/john/.local/lib/python3.7/site-packages/mu/logic.py", line 698, in check_usb
    def check_usb(self):
KeyboardInterrupt
2021-05-23 19:14:28,525 - mu.settings:169(save) DEBUG: Saving to /home/john/.local/share/mu/session.json
2021-05-23 19:14:28,528 - mu.settings:169(save) DEBUG: Saving to /home/john/.local/share/mu/venv.json

w4jkl, 2021-05-23:
Update:

I downloaded the Debian stable package for 1.0.2 and it installed into /usr/bin.
It ran OK.
After running 1.0.2, I then tried 1.1.0 beta 4 again and it runs OK.

abixadamj, 2021-05-24:
It looks like you press CTRL+C on keyboard during start process...

carlosperate, 2021-05-24:
Thanks for the report @w4jkl !

Looking at the log it seems like the last few attempts where interrupted by pressing Ctrl+C in the terminal that launched Mu.

Before that there seemed to be an issue installing some of the dependencies, probably caused by the dependency needing a relatively new version of pip (v19 if I remember correctly).

By installing mu from the Debian package it probably installed this dependency via packages as well (instead of fetching the Python package from PyPI, which caused the previous issue). That would explain why it works now.

Just to confirm this might be the case. Could you check what version of pip you are running?
```
pip --version
```
```
pip3 --version
```

w4jkl, 2021-05-24:
Here you go.

john@debian:~$ pip --version
bash: pip: command not found
john@debian:~$ pip3 --version
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
john@debian:~$

carlosperate, 2021-05-24:
Thank you @w4jkl!
That confirms my suspicion: https://github.com/mu-editor/mu/issues/1441#issuecomment-814790143
The cryptography module needs pip v19 and that's what caused this error captured in the log:
```
      =============================DEBUG ASSISTANCE=============================
      If you are seeing a compilation error please try the following steps to
      successfully install cryptography:
      1) Upgrade to the latest pip and try again. This will fix errors for most
         users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
      2) Read https://cryptography.io/en/latest/installation.html for specific
         instructions for your platform.
      3) Check our frequently asked questions for more information:
         https://cryptography.io/en/latest/faq.html
      4) Ensure you have a recent Rust toolchain installed:
         https://cryptography.io/en/latest/installation.html#rust
      5) If you are experiencing issues with Rust for *this release only* you may
         set the environment variable `CRYPTOGRAPHY_DONT_BUILD_RUST=1`.
      =============================DEBUG ASSISTANCE=============================
  error: can't find Rust compiler
  If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
  To update pip, run:
      pip install --upgrade pip
  and then retry package installation.
  If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
  This package requires Rust >=1.41.0.
  ----------------------------------------
  Failed building wheel for cryptography
  Running setup.py clean for cryptography
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-0fk1qr1b/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  running clean
  removing 'build/temp.linux-x86_64-3.7' (and everything under it)
  removing 'build/lib.linux-x86_64-3.7' (and everything under it)
  'build/bdist.linux-x86_64' does not exist -- can't clean it
  'build/scripts-3.7' does not exist -- can't clean it
  removing 'build'
  running clean_rust
      =============================DEBUG ASSISTANCE=============================
      If you are seeing a compilation error please try the following steps to
      successfully install cryptography:
      1) Upgrade to the latest pip and try again. This will fix errors for most
         users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
      2) Read https://cryptography.io/en/latest/installation.html for specific
         instructions for your platform.
      3) Check our frequently asked questions for more information:
         https://cryptography.io/en/latest/faq.html
      4) Ensure you have a recent Rust toolchain installed:
         https://cryptography.io/en/latest/installation.html#rust
      5) If you are experiencing issues with Rust for *this release only* you may
         set the environment variable `CRYPTOGRAPHY_DONT_BUILD_RUST=1`.
      =============================DEBUG ASSISTANCE=============================
  error: can't find Rust compiler
  If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
  To update pip, run:
      pip install --upgrade pip
  and then retry package installation.
  If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
  This package requires Rust >=1.41.0.
  ----------------------------------------
  Failed cleaning build dir for cryptography
Failed to build cryptography
ERROR: Failed to build one or more wheels
```

As this should be already resolved and ready for the next beta release I'll close this issue, but thank you very much @w4jkl for submitting this crash report and all the testing! Let us know if you find anything else 👍

closed 2021-05-24 by carlosperate