mu-editor/mu #1592: Crash Report

jbdblack, 2021-05-25:
[mu.log](https://github.com/mu-editor/mu/files/6536148/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\joryd\AppData\Local\python\mu\mu_venv-38-20210524-215010

```

carlosperate, 2021-07-05:
Thanks for the report and the log @jbdblack!

We've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-08-12:
Hi @jbdblack,

Just wondering if you were able to check if beta.5 works for you?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

jbdblack, 2021-08-21:
Hi Carlos,

Thanks for following up. I just tried installing Mu using the link you provided, and it appears to have crashed once again.

I can submit another log report, but at this point I've switched to a different editor. I just needed one to follow along with an introductory Python book.

Thank you,
Jory

https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmu-editor%2Fmu%2Freleases%2Ftag%2F1.1.0-beta.5&data=04%7C01%7Cjbb140030%40utdallas.edu%7Cecc7980ddf9a4234107a08d95d9022eb%7C8d281d1d9c4d4bf7b16e032d15de9f6c%7C0%7C0%7C637643695932924283%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=5U5eDl7gpEY4vcr3ID3O7yXDXuy%2BzS2wycrBkyKjMuM%3D&reserved=0>

carlosperate, 2021-08-23:
Hi Jory,

Yes please, if you could send the new crash report that would be really helpful for us to fix these issues.

Thanks!

carlosperate, 2021-12-17:
Thank you for sending the other crash report @jbdblack, it's really appreciated!
I've added a comment to https://github.com/mu-editor/mu/issues/1766#issuecomment-994747126, as I think we might have found a solution to the problem, would you be able to test the installer linked there?

I'll close this issue to continue the conversation over there, than you again for all your help identifying this problem!

closed 2021-12-17 by carlosperate