mu-editor/mu #1596: Could not find an attached device

jmlippincott, 2021-05-26:
# Bug Report

What were you trying to do?
---------------------------
Open Serial monitor or Plotter within Mu

What steps did you take to trigger the issue?
---------------------------------------------
Press 'Serial' or 'Plotter' menu button

What did you expect to happen?
------------------------------
Expected the built in serial monitor or plotter function to operate

What actually happened?
-----------------------
Error message: Could not find an attached device

Why is this difference problematic? (it might not be a bug!)
----------------------------------------------------------

Technical Info
--------------
* Mu version: 1.1.0.beta.4
* Operating system version: Mac OSX 11.3.1

Other Info
----------

- Device is Adafruit Circuit Playground Express running CircuitPython 6.2.0
- Mu Editor will load and save Code.py to device
- Serial Monitor can be started through Mac OSX Terminal; operates as expected

Editor Log
----------

[Mu Log.txt](https://github.com/mu-editor/mu/files/6546712/Mu.Log.txt)

jmlippincott, 2021-05-26:
That's the confusing piece here; the device is attached and Mu is able to edit/save the code.py from the device.  I've attached the most recent log, immediately following an attempt to open Mu Serial.  I've also attached a screenshot to show the REPL from the device, active in my Mac OSX terminal (iTerm 2 + oh-my-zsh), next to the window of Mu Editor showing the code.py loaded from the device, while also showing the "Could not find an attached device" error.

[image]

[Mu Log.txt](https://github.com/mu-editor/mu/files/6548313/Mu.Log.txt)

carlosperate, 2021-05-26:
Thanks for adding the new log with the extra info. This is the relevant part:

```
2021-05-26 12:27:06,147 - mu.interface.main:942(show_message) DEBUG: Could not find an attached device.
2021-05-26 12:27:06,147 - mu.interface.main:943(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2021-05-26 12:27:29,218 - mu.modes.base:474(toggle_repl) INFO: Toggle REPL on.
2021-05-26 12:27:32,072 - mu.interface.main:942(show_message) DEBUG: Could not find an attached device.
2021-05-26 12:27:32,073 - mu.interface.main:943(show_message) DEBUG: Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.
2021-05-26 12:27:35,169 - mu.modes.base:474(toggle_repl) INFO: Toggle REPL on.
```

So it's not seeing the serial device at all (I was hoping maybe it was a permission error).
By any chance are you running Mu on an M1 macOS?

jmlippincott, 2021-05-26:
I am, yes!  Sorry, didn't realize that was a relevant piece of information.
[image]

carlosperate, 2021-05-26:
Only relevant because it was a known issue with detecting the serial port in the M1: https://github.com/mu-editor/mu/issues/1543

Normally the CPU architecture is captured in the log, but because Mu runs with Rosetta it reports as x86_64.

One thing you can try to do is update the `adafruit_board_toolkit` dependency, which contains a fix. From the directory where Mu.app is installed (likely `/Applications`) you could try:

```
mu-editor.app/Contents/Resources/Python/bin/python3.8 -m pip install --upgrade adafruit_board_toolkit
```

jmlippincott, 2021-05-26:
Fixed!  Thanks so much!
[image]
[image]

closed 2021-05-26 by jmlippincott

carlosperate, 2021-05-26:
Awesome, glad you got it working!
Thanks so much for the report and all the testing, let us know if you find something else 👍

dhalbert, 2021-05-26:
Glad and relieved to see the fix works in real life.