mu-editor/mu #1597: Crash Report

RedLotusVenerable, 2021-05-26:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: de_DE
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-06-04:
Hi @RedLotusVenerable,  just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\Kevin\AppData\Local\python\mu\logs`.

RedLotusVenerable, 2021-06-08:
2021-05-26 16:35:52,892 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-26 16:35:52,892 - root:266(run) INFO: uname_result(system='Windows', node='Sophie', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')
2021-05-26 16:35:52,893 - root:267(run) INFO: Python path: ['C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-26 16:35:52,893 - root:268(run) INFO: Language code: de_DE
2021-05-26 16:35:52,893 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Kevin\AppData\Local\python\mu\settings.json; skipping
2021-05-26 16:35:52,893 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Kevin\AppData\Local\python\mu\session.json; skipping
2021-05-26 16:35:52,896 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-26 16:35:52,896 - root:266(run) INFO: uname_result(system='Windows', node='Sophie', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')
2021-05-26 16:35:52,896 - root:267(run) INFO: Python path: ['C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Kevin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-26 16:35:52,896 - root:268(run) INFO: Language code: de_DE
2021-05-26 16:35:52,897 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Kevin\AppData\Local\python\mu\settings.json; skipping
2021-05-26 16:35:52,897 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Kevin\AppData\Local\python\mu\session.json; skipping
2021-05-26 16:35:53,277 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-26 16:35:53,740 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-26 16:35:53,740 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539
2021-05-26 16:35:53,740 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-26 16:35:53,740 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539
2021-05-26 16:35:53,755 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210526-163539
2021-05-26 16:35:57,530 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: RuntimeError: failed to build image wheel, setuptools, pip because:
Traceback (most recent call last):
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 42, in install
    self._sync(filename, into)
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\copy.py", line 13, in _sync
    copy(src, dst)
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path\_sync.py", line 53, in copy
    method(norm(src), norm(dest))
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path\_sync.py", line 64, in copytree
    shutil.copy(src_f, dest_f)
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 415, in copy
[... cut ...]
    self.ensure_key_modules()
  File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
2021-05-26 16:36:21,132 - mu.settings:169(save) DEBUG: Saving to C:\Users\Kevin\AppData\Local\python\mu\session.json
2021-05-26 16:36:21,133 - mu.settings:169(save) DEBUG: Saving to C:\Users\Kevin\AppData\Local\python\mu\venv.json

carlosperate, 2021-06-08:
Thank you @RedLotusVenerable, that is indeed very helpful!

The interesting part of the log:
```
2021-05-26 16:35:53,755 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210526-163539
2021-05-26 16:35:57,530 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: RuntimeError: failed to build image wheel, setuptools, pip because:
Traceback (most recent call last):
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
installer.install(creator.interpreter.version_info)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 42, in install
self._sync(filename, into)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\copy.py", line 13, in _sync
copy(src, dst)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path_sync.py", line 53, in copy
method(norm(src), norm(dest))
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path_sync.py", line 64, in copytree
shutil.copy(src_f, dest_f)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 415, in copy
copyfile(src, dst, follow_symlinks=follow_symlinks)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 261, in copyfile
with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539\Lib\site-packages\wheel\cli\unpack.py'
Traceback (most recent call last):
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
installer.install(creator.interpreter.version_info)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 42, in install
self._sync(filename, into)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\copy.py", line 13, in _sync
copy(src, dst)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path_sync.py", line 53, in copy
method(norm(src), norm(dest))
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path_sync.py", line 64, in copytree
shutil.copy(src_f, dest_f)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 416, in copy
copymode(src, dst, follow_symlinks=follow_symlinks)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 305, in copymode
chmod_func(dst, stat.S_IMODE(st.st_mode))
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden: 'C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539\Lib\site-packages\pkg_resources\_vendor\appdirs.py'
Traceback (most recent call last):
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
installer.install(creator.interpreter.version_info)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 42, in install
self._sync(filename, into)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\copy.py", line 13, in _sync
copy(src, dst)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path_sync.py", line 53, in copy
method(norm(src), norm(dest))
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path_sync.py", line 64, in copytree
shutil.copy(src_f, dest_f)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 416, in copy
copymode(src, dst, follow_symlinks=follow_symlinks)
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 305, in copymode
chmod_func(dst, stat.S_IMODE(st.st_mode))
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden: 'C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539\Lib\site-packages\pip\_internal\commands\cache.py'
```

So we end up with:

```
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 261, in copyfile
with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539\Lib\site-packages\wheel\cli\unpack.py'
```

```
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 305, in copymode
chmod_func(dst, stat.S_IMODE(st.st_mode))
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden: 'C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539\Lib\site-packages\pkg_resources\_vendor\appdirs.py'
```

```
File "C:\Users\Kevin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 305, in copymode
chmod_func(dst, stat.S_IMODE(st.st_mode))
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden: 'C:\Users\Kevin\AppData\Local\python\mu\mu_venv-38-20210526-163539\Lib\site-packages\pip\_internal\commands\cache.py'
```

Is it possible you have antivirus running that might be affecting Mu operation?
If you do, could you try disabling it, opening Mu and seeing if that works? If it doesn't a new copy of the log file after it fails again would be really useful to contrast the error messages.

Thanks!

carlosperate, 2021-08-05:
Hi @RedLotusVenerable, I was wondering if it was possible you had an antivirus running while starting up Mu.

If so, could you try disabling it, opening Mu and seeing if that works? If it doesn't a new copy of the log file after it fails again would be really useful to contrast the error messages.

Thanks!

carlosperate, 2021-12-17:
Hi @RedLotusVenerable, are you still experiencing this problem?

Is it possible you had an antivirus running while starting up Mu?
If so, could you try disabling it, opening Mu and seeing if that works? If it doesn't a new copy of the log file after it fails again would be really useful to contrast the error messages.

Thanks!

carlosperate, 2022-05-31:
Closing due to lack of activity.

closed 2022-05-31 by carlosperate