mu-editor/mu #1599: Crash Report starting mu - Unable to create a virtual environment

randomsmith9, 2021-05-27:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_AU
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-213849

```

randomsmith9, 2021-05-27:
mu.log
2021-05-27 21:45:12,071 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-27 21:45:12,072 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-2FP51K9', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2021-05-27 21:45:12,072 - root:267(run) INFO: Python path: ['C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\smith\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\smith\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-27 21:45:12,075 - root:268(run) INFO: Language code: en_AU
2021-05-27 21:45:12,075 - mu.settings:220(load) WARNING: No settings file found at C:\Users\smith\AppData\Local\python\mu\settings.json; skipping
2021-05-27 21:45:12,213 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-27 21:45:12,217 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-27 21:45:12,217 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214257
2021-05-27 21:45:12,217 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210527-214257
2021-05-27 21:45:12,684 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: AttributeError: module 'virtualenv.create.via_global_ref.builtin.cpython.mac_os' has no attribute 'CPython3macOsFramework'
2021-05-27 21:45:12,684 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214257 as C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214257.FAILED
Traceback (most recent call last):
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214257

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\smith\\AppData\\Local\\python\\mu\\mu_venv-38-20210527-214257' -> 'C:\\Users\\smith\\AppData\\Local\\python\\mu\\mu_venv-38-20210527-214257.FAILED'
2021-05-27 21:45:12,700 - mu.virtual_environment:410(relocate) DEBUG: Virtual environment set up mu_venv-38-20210527-214512 at C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214512
2021-05-27 21:45:12,700 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x00000224D8586EE0>; Splash Handler; <SplashLogHandler (INFO)>
2021-05-27 21:45:12,700 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #2.
2021-05-27 21:45:12,700 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214512
2021-05-27 21:45:12,700 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210527-214512
2021-05-27 21:45:12,949 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: AttributeError: module 'virtualenv.create.via_global_ref.builtin.cpython.mac_os' has no attribute 'CPython3macOsFramework'
2021-05-27 21:45:12,949 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214512 as C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214512.FAILED
Traceback (most recent call last):
  File "C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
[... cut ...]
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\smith\AppData\Local\python\mu\mu_venv-38-20210527-214512
2021-05-27 21:45:20,267 - mu.settings:169(save) DEBUG: Saving to C:\Users\smith\AppData\Local\python\mu\session.json
2021-05-27 21:45:20,268 - mu.settings:169(save) DEBUG: Saving to C:\Users\smith\AppData\Local\python\mu\venv.json

carlosperate, 2021-05-27:
Thank you for the report and the log @randomsmith9!

Based on this error, it seems like it might be a bad installation of the virtualenv package: https://github.com/pypa/virtualenv/issues/1740
```
EBUG: Process returned 1; output: AttributeError: module 'virtualenv.create.via_global_ref.builtin.cpython.mac_os' has no attribute 'CPython3macOsFramework'
```

How did you install Mu, via installer or pip? Could you try uninstalling and installing again?

randomsmith9, 2021-05-29:
I installed via the windows installer.
tried uninstalling and installing again, no luck
tried uninstalling virtualenv and then following the virtual env install instructions from link you posted but that didn't help
https://github.com/pypa/virtualenv/issues/1740#issuecomment-603623727

randomsmith9, 2021-05-29:
Ran command 
"C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip update virtualenv
that seems to have fixed it an upgraded virtual env
"C:\Users\smith\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list
to
20.4.7
can launch the gui now
thanks for the link and you'll have the commands for others to try on windows if they have the same issue
cheers
David

closed 2021-05-29 by randomsmith9

carlosperate, 2021-05-31:
I'm really glad you were able to resolve this, and thanks for sharing the commands you've used to fix it!