mu-editor/mu #1600: venv pip install step does not install all wheels

carlosperate, 2021-05-27:
## Issue description
We've had a series of Crash reports where it looks like `pip install ...` step for wheels finishes successfully, but then the `pip list` only shows a partial list of packages.

These problems are usually manifested with the `ModuleNotFoundError: No module named 'pgzero'` (or a different module), as it might have installed some of the wheels, but not reached `pgzero`.

## Example log
For example in https://github.com/mu-editor/mu/issues/1591 we find:

```
2021-05-24 14:48:24,370 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cryptography-3.4.7-cp36-abi3-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\decorator-5.0.7-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ecdsa-0.16.1-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\esptool-3.0-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\flask-1.1.2-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipykernel-5.5.3-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython-7.22.0-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython_genutils-0.2.0-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\itsdangerous-1.1.0-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jedi-0.18.0-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jinja2-2.11.3-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_client-6.1.12-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_core-4.7.1-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\markupsafe-1.1.1-cp38-cp38-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\numpy-1.20.2-cp38-cp38-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\parso-0.8.2-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pgzero-1.2.1-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pickleshare-0.7.5-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\prompt_toolkit-3.0.18-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pycparser-2.20-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pygame-2.0.1-cp38-cp38-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pygments-2.8.1-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pyserial-3.5-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\python_dateutil-2.8.1-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pywin32-300-cp38-cp38-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pyzmq-22.0.3-cp38-cp38-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\qtconsole-4.7.4-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\qtpy-1.9.0-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\reedsolo-1.5.4-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\setuptools-56.0.0-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\six-1.15.0-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\tornado-6.1-cp38-cp38-win_amd64.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\traitlets-5.0.5-py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\wcwidth-0.2.5-py2.py3-none-any.whl
Processing c:\users\zachary.desktop-96jpn9t\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\werkzeug-1.0.1-py2.py3-none-any.whl
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, pywin32, parso, tornado, setuptools, pyzmq, python-dateutil, Pygments, pycparser, prompt-toolkit, pickleshare, jupyter-core, jedi, decorator, colorama, backcall, MarkupSafe, jupyter-client, ipython, cffi, Werkzeug, reedsolo, QtPy, pyserial, pygame, numpy, Jinja2, itsdangerous, ipykernel, ecdsa, cryptography, click, bitstring, qtconsole, pgzero, Flask, esptool, adafruit-board-toolkit
  Attempting uninstall: setuptools
    Found existing installation: setuptools 54.1.2
    Uninstalling setuptools-54.1.2:
      Successfully uninstalled setuptools-54.1.2
2021-05-24 14:48:26,649 - mu.virtual_environment:193(run) DEBUG: Process output: Package          Version
---------------- -------
backcall         0.2.0
colorama         0.4.4
decorator        5.0.7
ipython          7.22.0
ipython-genutils 0.2.0
jedi             0.18.0
jupyter-client   6.1.12
jupyter-core     4.7.1
MarkupSafe       1.1.1
parso            0.8.2
pickleshare      0.7.5
pip              21.1.2
prompt-toolkit   3.0.18
pycparser        2.20
Pygments         2.8.1
python-dateutil  2.8.1
pywin32          300
pyzmq            22.0.3
setuptools       56.0.0
six              1.15.0
tornado          6.1
traitlets        5.0.5
wcwidth          0.2.5
wheel            0.36.2
```
So it in this cased it failed at `cffi` as every wheels listed before it was installed correctly and none of the wheels installed after it are shown in the pip list.

## Current theory
Our assumption right now is that perhaps pip outputs an error in the stderr pipeline we are not capturing correctly in the log.
However, @tjguk has already looked into this and technically speaking it's all configured correctly. PR https://github.com/mu-editor/mu/pull/1530 reworks this slightly to see if a different method is capable to capture the missing output.

carlosperate, 2021-05-27:
Merging https://github.com/mu-editor/mu/issues/1591 here:

> [mu.log](https://github.com/mu-editor/mu/files/6534686/mu.log)
> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Windows 10 10.0.10240 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Zachary.DESKTOP-96JPN9T\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\Zachary.DESKTOP-96JPN9T\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\Zachary.DESKTOP-96JPN9T\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
>     self.ensure()
>   File "C:\Users\Zachary.DESKTOP-96JPN9T\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
>     self.ensure_key_modules()
>   File "C:\Users\Zachary.DESKTOP-96JPN9T\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
>     raise VirtualEnvironmentEnsureError(
> mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
> ```

All the errors in this log file are related to this issue. Each time stopping at a different package: `pyserial`, `cffi`, `cffi`,  `click`.

carlosperate, 2021-05-27:
Merging https://github.com/mu-editor/mu/issues/1579 here:
> [mu.log](https://github.com/mu-editor/mu/files/6510379/mu.log)
> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\nissim.tushiah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\nissim.tushiah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\nissim.tushiah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
>     self.ensure()
>   File "C:\Users\nissim.tushiah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
>     self.ensure_key_modules()
>   File "C:\Users\nissim.tushiah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
>     raise VirtualEnvironmentEnsureError(
> mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
> ```

All the errors in this log file are related to this issue. Each time stopping at a different package: `Jinja2`, `click`

carlosperate, 2021-05-27:
Not merging here, as this issue contains other problem, but one of them is the same as this: https://github.com/mu-editor/mu/issues/1559

The other problems might not be related, so we'll continue debugging them in the original ticket.

carlosperate, 2021-05-27:
Merging https://github.com/mu-editor/mu/issues/1574 here:

> [mu.log](https://github.com/mu-editor/mu/files/6496560/mu.log)
> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_GB
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
>     self.ensure()
>   File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
>     self.ensure_key_modules()
>   File "C:\Users\Luke\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
>     raise VirtualEnvironmentEnsureError(
> mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
> ```

All the errors in this log file are related to this issue.

carlosperate, 2021-05-31:
Merging https://github.com/mu-editor/mu/issues/1605 here:

> [mu.log](https://github.com/mu-editor/mu/files/6563926/mu.log)
> 
> **REMEMBER TO ATTACH YOUR LOG FILE**
>
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\DELL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\DELL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\DELL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
>     self.ensure()
>   File "C:\Users\DELL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
>     self.ensure_key_modules()
>   File "C:\Users\DELL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
>     raise VirtualEnvironmentEnsureError(
> mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
> ```

All the errors in this log file are related to this issue. Each time stopping at a different package: `cffi`, `numpy`, `cffi`, `numpy`.

carlosperate, 2021-05-31:
I think maybe the problem we are seeing is a timeout of the pip install step from:
https://github.com/mu-editor/mu/blob/de20993f82cf5b3e7c9d9efcfc282d83c6c427f7/mu/virtual_environment.py#L210-L217

If we separate each wheel into its own `pip install` it should resolve this possible problem and provide more granular info in the logs.

tjguk, 2021-05-31:
I'll make that change in #1530.

carlosperate, 2021-05-31:
Okay, by running Mu in a virtualised environment and also compiling a bunch of things in the background to consume CPU I was able to recreate this problem (stopping at cffi), and also noticed this was printed into the console:

```
QProcess: Destroyed while process ("/home/pi/.local/share/mu/mu_venv-35-20210527-122405/bin/pip") is still running.
```

carlosperate, 2021-05-31:
Running this PR https://github.com/mu-editor/mu/pull/1610/ in the same environment as my previous test and the problem doesn't manifest any more, so hopefully that should fix it.

@tjguk I think perhaps the issue about not getting all of the pip stderr might just have been the process being killed, which is not captured in the logs, and thinking the process ended without the error being captured.

tjguk, 2021-05-31:
#1610 looks good, @carlosperate . I was converging on sthg similar using --no-inded and --find-links but if yours works equivalently please go ahead and merge once you're happy with tests. I'll back that element of my change out. 

I think with that change plus the corrected error-trapping we're in a much better place.

We still have the problem in #1421  where a later installation can leave us with a mixed bag of wheels. I did start a branch to parse the wheel filenames and discard all but the latest; not sure what happened to that...

carlosperate, 2021-05-31:
Thanks Tim!

> We still have the problem in #1421 where a later installation can leave us with a mixed bag of wheels. I did start a branch to parse the wheel filenames and discard all but the latest; not sure what happened to that...

Yeah, I was thinking if we should use the ` --force-reinstall` or `--ignore-installed` flags, but that might cause problems that are harder to debug later on, so might be better to fail early at the venv creation step.

carlosperate, 2021-06-09:
Merging https://github.com/mu-editor/mu/issues/1628 here:

> [mu.log](https://github.com/mu-editor/mu/files/6624547/mu.log)
> 
> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Windows 10 10.0.18362 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Eric\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\Eric\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\Eric\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
>     self.ensure()
>   File "C:\Users\Eric\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
>     self.ensure_key_modules()
>   File "C:\Users\Eric\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
>     raise VirtualEnvironmentEnsureError(
> mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
> ```

All the errors in this log file are related to this issue. Each time stopping at a different package: cffi, jedi, cffi, numpy, pygame,

muttstang, 2021-06-09:
Any idea when the next release with this fixed is to be released?

carlosperate, 2021-06-10:
Hopefully "soon", but we don't have any dates, sorry. We are all volunteers working on this in our free time, so we can't really promise any timelines.
I'll update this ticket when the latest release is out for people to test it it.

In the meantime, if you are feeling adventurous, you can try an installer from the CI (in the artefacts section in https://github.com/mu-editor/mu/actions/runs/894156733). These are not official releases in any way, just test builds for development, but might be okay to use until the next official beta release.

carlosperate, 2021-06-27:
Merging https://github.com/mu-editor/mu/issues/1666 here:

> Mu Crash Report
> Something has gone wrong with Mu. Here's the error:
> ```
> Traceback (most recent call last):
> File "C:\Users\Ann\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
> raise ex
> File "C:\Users\Ann\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
> venv.ensure_and_create(self.display_text)
> File "C:\Users\Ann\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
> self.ensure()
> File "C:\Users\Ann\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
> self.ensure_key_modules()
> File "C:\Users\Ann\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
> raise VirtualEnvironmentEnsureError(
> mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
> ```

carlosperate, 2021-07-09:
@muttstang @zrose49 @wagonis @Alice-star-lgtm @lpoolehfed @xtephenz @ AnnEvans 

Hi everyone!

We believe the issues you've had with Mu might be fixed in beta.5. Could you install this version and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thank you!

AnnEvans, 2021-07-10:
Hi Carlos

Despite many efforts I can't get this to download. Microsoft Defender blocks the msi file as unsafe (so I can't even try to install it). I have reported it as a safe file so hopefully this will be resolved quickly.

Thanks & fingers crossed!

https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-08-04:
Oh no! Thanks for trying @AnnEvans!

We have this ticket discussing possible issues with antivirus so if it's okay let's move the conversation over there: https://github.com/mu-editor/mu/issues/1495#issuecomment-892781463

carlosperate, 2021-09-22:
We haven't received any reports about this issue since beta.5 was released, so I'll consider this resolved.
Thanks everybody!

closed 2021-09-22 by carlosperate