mu-editor/mu #1601: Crash Report

donwatkins, 2021-05-27:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 5.4.0-73-generic #82-Ubuntu SMP Wed Apr 14 17:39:42 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/don/.local/lib/python3.8/site-packages/mu/logic.py", line 698, in check_usb
    def check_usb(self):
KeyboardInterrupt

```

ajs256, 2021-05-27:
This is okay, and not a crash. See #1547.

carlosperate, 2021-07-05:
PR https://github.com/mu-editor/mu/pull/1625 has been merged and has been included in the next beta.5 release, thanks @ajs256, and thanks @donwatkins for the report! 
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

closed 2021-07-05 by carlosperate