mu-editor/mu #1602: Mu 1.1.0-beta-4 wont run at all

electric1236, 2021-05-27:
I was previously using Mu 1.0.3 and ran into some glitches with a teensy 4.1 so I uninstalled it and tried installing the latest version following all the instructions and after it was installed I couldn't open it. When I launched for about 3 seconds I had a blue spinning loading thing and then it goes away and nothing comes on screen. I tried on another computer and had the same problem. Does anyone know what is wrong?

Im using:
Windows 10 home, 20H2, and the other computer I tested on was windows 10 as well though hasnt been update in a few months, I used the windows 64 bit installer.

I have tried:
restarting my computer multiple times, uninstalling it and reinstalling it multiplytimes, disabling Malwarebytes, updating to the latest windows 10, running it using administrator, I also tried 1.1.0-beta.3 and 1.1.0-beta.2 and a few other versions (only 1.0.2 and 1.0.3 worked though I didn't try all of them), I looked for it in task manager and didn't see it running.

I checked Windows: %LOCALAPPDATA%\python\mu\Logs for logs but there were no logs.

thank you for your help

mitag, 2021-05-28:
I seem to have the same problem here, Mu crashes on startup. I have been using Mu since the mu-editor_rc1_win64 version in 2018 on this system and never had any problems. 

[mu.log](https://github.com/mu-editor/mu/files/6561287/mu.log)

electric1236, 2021-05-28:
I'm not sure if it is crashing for me as I don't see anything appear on the screen (and I don't get any logs), do you see anything appear on screen?

mitag, 2021-05-28:
You are right, I could see something before the crash and delete my report.

carlosperate, 2021-08-24:
Hi @electric1236, are you still experiencing this problem?

What installation instructions did you follow? Were you using the installers or pip?

Can you execute this in a command prompt and see what messages it outputs?

```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m mu
```

electric1236, 2021-08-25:
I tried beta-5 out and it didn't work either. I ran the command the first time it did nothing the second time it get trace back error.
[image]

carlosperate, 2021-08-25:
Thank you very much for the additional information @electric1236, that is really useful!

Funnily enough, when googling the error one of the first results is a thread from this issue tracker 😅 https://github.com/mu-editor/mu/issues/354

In one of the two computer you tried this, would it be possible to make sure all Windows updates, including optional updates, are installed?
The similar issue we had before was on Windows versions below 10, but it would be really good if we could inmediately discard the issue being a missing update.

mitag, 2021-09-01:
Did you manually uninstall the versions with the .exe installer? The newer versions with the .msi installer only recognise the .msi versions and can only uninstall these automatically.

See also here: [https://github.com/mu-editor/mu/issues/1604#issuecomment-851789394](url)

electric1236, 2021-09-01:
yes, normally when I test it I uninstall every version before installing the version im testing. I uninstall them normally with the control panel, though I have tried windows settings to uninstall. 

I uninstalled 1.1.0-alpha using the uninstall.exe in its directory then confirmed in the control panel I didn't have any other mu versions installed. Then I downloaded 1.1.0b5 and installed it and tried to run it and it didn't run, so I uninstalled it by using the .msi installer file I downloaded. Then I used that same file to reinstall it, and it didn't work again, I got the same error msg in the cmd when I ran that command again from above.

the attached photo of after I did completed all the steps mentioned in the 2nd paragraph
[image]

carlosperate, 2021-09-01:
Thank you for your perseverance @electric1236 and @mitag!

I wonder if PyQt5 is being loaded from the wrong place. Could you run the following commands and paste the results here?

```
echo %PATH%
```
```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"
```
```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

Thanks again with all your help debugging this issue!

electric1236, 2021-09-04:
Thank you for all your help

I ran this on my desktop PC, I try it on my laptop later today.

I ran echo %PATH%
`C:\Users\Bens Pc>echo %PATH%                                                                                                   C:\WINDOWS;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\HMITool6.0\Marco\HMI\bin;C:\Program Files (x86)\HMITool6.0\Marco\X86\bin;%SystemRoot%\system32;%SystemRoot%;%SystemRoot%\System32\Wbem;%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\;%SYSTEMROOT%\System32\OpenSSH\;C:\Users\Bens Pc\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Nmap;`

running "%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"
`C:\Users\Bens Pc>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"                        ['', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Bens Pc\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']`

running "%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
`C:\Users\Bens Pc>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose                                                                                                                                                                                                                                 Package                Version      Location                                                                                               Installer                                                                                                                                                                         ---------------------- ------------ ------------------------------------------------------------------------------------------------------ ---------                                                                                                                                                                         adafruit-board-toolkit 1.1.0        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                                                                                                               appdirs                1.4.4        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                                                                                                               backcall               0.2.0        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                                                                                                               black                  21.6b0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                                                                                                               click                  8.0.1        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                                                                                                               colorama               0.4.4        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                 [... cut ...] 296323 (from https://pypi.org/simple/pip/) (requires-python:>=3.6), version: 21.2.4                       Found link https://files.pythonhosted.org/packages/52/e1/06c018197d8151383f66ebf6979d951995cf495629fc54149491f5d157d0/pip-21.2.4.tar.gz#sha256=0eb8a1516c3d138ae8689c0c1a60fde7143310832f9dc77e11d8a4bc62de193b (from https://pypi.org/simple/pip/) (requires-python:>=3.6), version: 21.2.4                               Given no hashes to check 177 links for project 'pip': discarding no candidates`

electric1236, 2021-09-05:
I ran it on my laptop and got

`C:\Users\Benjamin Whiffen>echo %PATH%                                                                                   C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\Program Files (x86)\Razer\ChromaBroadcast\bin;C:\Program Files\Razer\ChromaBroadcast\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\HMITool6.0\Marco\HMI\bin;C:\Program Files (x86)\HMITool6.0\Marco\X86\bin;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Windows\system32\config\systemprofile\AppData\Local\Microsoft\WindowsApps;C:\Program Files\TortoiseSVN\bin;C:\Program Files (x86)\dotnet\;C:\Users\Benjamin Whiffen\AppData\Local\Microsoft\WindowsApps;                                                                                                                                                                                                    C:\Users\Benjamin Whiffen>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"        ['', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Benjamin Whiffen\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Benjamin Whiffen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']                                                                                                                                C:\Users\Benjamin Whiffen>                                                                                                                                                                                                                      C:\Users\Benjamin Whiffen>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose                   Package                Version      Location                                                                                                        Installer                                                                                   ---------------------- ------------ --------------------------------------------------------------------------------------------------------------- ---------                                                                                   adafruit-board-toolkit 1.1.0        c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                         appdirs                1.4.4        c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                         backcall               0.2.0        c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                         black                  21.6b0       c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                         click                  8.0.1        c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                         colorama               0.4.4        c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                         debugpy                1.3.0        c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                         decorator              5.0.9        c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                      [... cut ...] 38ae8689c0c1a60fde7143310832f9dc77e11d8a4bc62de193b (from https://pypi.org/simple/pip/) (requires-python:>=3.6), version: 21.2.4                                                                          Given no hashes to check 177 links for project 'pip': discarding no candidates                                                                                                                                                                  C:\Users\Benjamin Whiffen>                               `

carlosperate, 2021-09-06:
Thanks @electric1236!

I think the issue is that something is causing Mu to pick up the PyQt packages from a different location than it's own site-pacakges.

In your PC pip output we can confirm it is taking PyQt pacakges from `appdata\roaming\python\python38\site-packages` instead of `appdata\local\programs\mu editor\python\lib\site-packages`:
```
Package                Version      Location                                                                     Installer
---------------------- ------------ ---------------------------------------------------------------------------- ---------
PyQt5                  5.15.2       c:\users\bens pc\appdata\roaming\python\python38\site-packages               pip
pyqt5-plugins          5.15.2.2.0.1 c:\users\bens pc\appdata\roaming\python\python38\site-packages               pip
PyQt5-sip              12.9.0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages   pip
pyqt5-tools            5.15.2.3     c:\users\bens pc\appdata\roaming\python\python38\site-packages               pip
PyQtChart              5.13.1       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages   pip
QScintilla             2.11.3       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages   pip
qt5-applications       5.15.2.2.1   c:\users\bens pc\appdata\roaming\python\python38\site-packages               pip
qt5-tools              5.15.2.1.0.1 c:\users\bens pc\appdata\roaming\python\python38\site-packages               pip
```
Windows path does not have it:
```
C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common
C:\Program Files (x86)\Common Files\Oracle\Java\javapath
C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler
C:\ProgramData\Oracle\Java\javapath
C:\Program Files (x86)\Razer Chroma SDK\bin
C:\Program Files\Razer Chroma SDK\bin
C:\Windows\system32
C:\Windows
C:\Windows\System32\Wbem
C:\Windows\System32\WindowsPowerShell\v1.0\
C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\
C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\
C:\Program Files\Microsoft SQL Server\120\Tools\Binn\
C:\Program Files\Microsoft SQL Server\120\DTS\Binn\
C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\
C:\Program Files\Microsoft SQL Server\110\Tools\Binn\
C:\WINDOWS\system32
C:\WINDOWS
C:\WINDOWS\System32\Wbem
C:\WINDOWS\System32\WindowsPowerShell\v1.0\
C:\WINDOWS\System32\OpenSSH\
C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR
C:\Program Files (x86)\HMITool6.0\Marco\HMI\bin
C:\Program Files (x86)\HMITool6.0\Marco\X86\bin
%SystemRoot%\system32
%SystemRoot%
%SystemRoot%\System32\Wbem
%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\
%SYSTEMROOT%\System32\OpenSSH\
C:\Users\Bens Pc\AppData\Local\Microsoft\WindowsApps
C:\Program Files (x86)\Nmap
```
But the Python path does:
```
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python'
'C:\\Users\\Bens Pc\\AppData\\Roaming\\Python\\Python38\\site-packages'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib'
'C:\\Users\\Bens Pc\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin'
```

And we can see the same thing in the laptop:
```
Package                Version      Location                                                                             Installer
---------------------- ------------ ------------------------------------------------------------------------------------ ---------
PyQt5                  5.15.2       c:\users\benjamin whiffen\appdata\roaming\python\python38\site-packages              pip 
pyqt5-plugins          5.15.2.2.0.1 c:\users\benjamin whiffen\appdata\roaming\python\python38\site-packages              pip 
PyQt5-sip              12.9.0       c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages  pip 
pyqt5-tools            5.15.2.3     c:\users\benjamin whiffen\appdata\roaming\python\python38\site-packages              pip 
PyQtChart              5.13.1       c:\users\benjamin whiffen\appdata\local\programs\mu editor\python\lib\site-packages  pip 
PyQtWebEngine          5.15.2       c:\users\benjamin whiffen\appdata\roaming\python\python38\site-packages              pip 
qt5-applications       5.15.2.2.1   c:\users\benjamin whiffen\appdata\roaming\python\python38\site-packages              pip 
qt5-tools              5.15.2.1.0.1 c:\users\benjamin whiffen\appdata\roaming\python\python38\site-packages              pip 
```
So, something is making it pick up `C:\Users\xxxxx\AppData\Roaming\Python\Python38\site-packages` as a place to look for packages. 🤔

electric1236, 2021-09-07:
I uninstall QT5 but it didn't seem to change anything.

Also both my computer had QT5 installed and anaconda with spyder and pycharm installed

tmontes, 2021-11-06:
All, I know that I AM VERY LATE to this party. A few notes:

* I edited @electric1236 comment with detailed output for readability (thanks for the detail!)
  (hint: next time, wrap your outputs in isolated triple-backtick lines, which is what I did)

* (for my future self) Interesting that `sys.path` includes a path foreign to Mu's install: `'C:\\Users\\Bens Pc\\AppData\\Roaming\\Python\\Python38\\site-packages'`. Intuition says its [this `pup` bug](https://github.com/mu-editor/pup/issues/42).

I'll be back!

carlosperate, 2021-12-15:
We think we might have found a solution thanks @hmmkay! Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

electric1236, 2021-12-23:
Hi, so I tried running it on my desktop and it didnt work.

I uninstalled the previous version restarted and then installed the version you linked to, and it did run, to I tried the command given above and got the same error as before.
```
Microsoft Windows [Version 10.0.19044.1415]                                                                             (c) Microsoft Corporation. All rights reserved.                                                                                                                                                                                                 C:\Users\Bens Pc>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m mu                                            Traceback (most recent call last):                                                                                        File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main            return _run_code(code, main_globals, None,                                                                            File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code                       exec(code, run_globals)                                                                                               File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 1, in <module>     from mu.app import run                                                                                                File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 46, in <module>         from .interface import Window                                                                                         File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\__init__.py", line 1, in <module>                                                                                                                   from .main import Window                                                                                              File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\main.py", line 65, in <module>                                                                                                                      from mu.interface.editor import EditorPane                                                                            File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\editor.py", line 25, in <module>                                                                                                                    from PyQt5.Qsci import (                                                                                            ImportError: DLL load failed while importing Qsci: The specified procedure could not be found.                                                                                                                                                  C:\Users\Bens Pc> 
```

I also tried the beta 7 version on the website and it didn't launch either

carlosperate, 2021-12-24:
Thanks @electric1236. Beta 7 was released a couple of days ago, could you uninstall that version of Mu and install this one?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.7

I suspect the result would be the same, but it's better to make sure we are testing the latest version to ensure it include other related bug fixes.

carlosperate, 2021-12-24:
After installing beta.7 could you run these two command again, thanks!

```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m mu
```

electric1236, 2021-12-24:
I installed the version you linked to and ran the commands and it didn't seem to work.

```Microsoft Windows [Version 10.0.19044.1415]                                                                             (c) Microsoft Corporation. All rights reserved.                                                                                                                                                                                                 C:\Users\Bens Pc>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose                            Package                           Version      Location                                                                                               Installer                                                                                 --------------------------------- ------------ ------------------------------------------------------------------------------------------------------ ---------                                                                                 adafruit-board-toolkit            1.1.0        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       appdirs                           1.4.4        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       backcall                          0.2.0        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       backports.entry-points-selectable 1.1.1        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       black                             21.12b0      c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       click                             8.0.3        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       colorama                          0.4.4        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       decorator                         5.1.0        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       distlib                           0.3.4        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       filelock                          3.4.0        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       flake8                            4.0.1        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       ipykernel                         5.5.6        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       ipython                           7.30.1       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       ipython-genutils                  0.2.0        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       jedi                              0.18.1       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       jupyter-client                    6.1.12       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       jupyter-core                      4.9.1        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       matplotlib-inline                 0.1.3        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip                                                                                       mccabe                            0.6.1        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages         [... cut ...] dule>                                                                                                                    from PyQt5.Qsci import (                                                                                            ImportError: DLL load failed while importing Qsci: The specified procedure could not be found.                                                                                                                                                  C:\Users\Bens Pc>   ```

carlosperate, 2021-12-24:
Thanks @electric1236

The interesting part of pip list is this:

```
C:\Users\Bens Pc>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
Package                           Version      Location                                                                                               Installer
--------------------------------- ------------ ------------------------------------------------------------------------------------------------------ ---------
...
mu-editor                         1.1.0b7      c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages
...
PyQt5                             5.15.2       c:\users\bens pc\appdata\roaming\python\python38\site-packages                                         pip
pyqt5-plugins                     5.15.2.2.0.1 c:\users\bens pc\appdata\roaming\python\python38\site-packages                                         pip
PyQt5-sip                         12.9.0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pyqt5-tools                       5.15.2.3     c:\users\bens pc\appdata\roaming\python\python38\site-packages                                         pip
PyQtChart                         5.13.1       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pyserial                          3.5          c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
python-dateutil                   2.8.2        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
python-dotenv                     0.15.0       c:\users\bens pc\appdata\roaming\python\python38\site-packages                                         pip
pywin32                           303          c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pyzmq                             22.3.0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
QScintilla                        2.11.3       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
qt5-applications                  5.15.2.2.1   c:\users\bens pc\appdata\roaming\python\python38\site-packages                                         pip
qt5-tools                         5.15.2.1.0.1 c:\users\bens pc\appdata\roaming\python\python38\site-packages                                         pip
qtconsole                         4.7.7        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
QtPy                              1.11.3       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
semver                            2.13.0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
setuptools                        57.4.0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages\setuptools-57.4.0-py3.8.egg
six                               1.16.0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
tomli                             1.2.3        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
tornado                           6.1          c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
traitlets                         5.1.1        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
typing-extensions                 4.0.1        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
virtualenv                        20.10.0      c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
wcwidth                           0.2.5        c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
wheel                             0.37.0       c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
```
From the pip list output everything below `PyQt5` is not from the Mu's site-packages (don't know if that is just coincidence, or more than that).

And we get the same exception:
```> "%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m mu                                            
Traceback (most recent call last):                                                                                      
  File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main        
    return _run_code(code, main_globals, None,                                                                          
  File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code                   
    exec(code, run_globals)                                                                                             
  File "C:\Users\Bens Pc\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 1, in <module> 
[... cut ...]
@electric1236 could you run these commands? Maybe the `-I` flag isolates mu from the system Python as well:

```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -I -m pip list --verbose
```

```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -I -m mu
```

When pasting the output could you make sure the new lines are present? I don't know why your copy/paste comes back in a single line, but it's impossible to read as it is and I have to spend quite a bit of time trying to untangle it multiple lines.

electric1236, 2021-12-25:
When I ran the second command mu started up and worked! I think their new lines are present now, hope this helps, and thank you for all your help.

```
Microsoft Windows [Version 10.0.19044.1415]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Bens Pc>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -I -m pip list --verbose
Package                           Version Location                                                                                               Installer
--------------------------------- ------- ------------------------------------------------------------------------------------------------------ ---------
adafruit-board-toolkit            1.1.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
appdirs                           1.4.4   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
backcall                          0.2.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
backports.entry-points-selectable 1.1.1   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
black                             21.12b0 c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
click                             8.0.3   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
colorama                          0.4.4   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
decorator                         5.1.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
distlib                           0.3.4   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
filelock                          3.4.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
flake8                            4.0.1   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipykernel                         5.5.6   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipython                           7.30.1  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
ipython-genutils                  0.2.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jedi                              0.18.1  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jupyter-client                    6.1.12  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
jupyter-core                      4.9.1   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
matplotlib-inline                 0.1.3   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
mccabe                            0.6.1   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
mu-editor                         1.1.0b7 c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages
mypy-extensions                   0.4.3   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
nudatus                           0.0.5   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
parso                             0.8.3   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pathspec                          0.9.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pickleshare                       0.7.5   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pip                               21.1.3  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages\pip-21.1.3-py3.8.egg
platformdirs                      2.4.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
prompt-toolkit                    3.0.24  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pycodestyle                       2.8.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
pyflakes                          2.4.0   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
Pygments                          2.10.0  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
PyQt5                             5.13.2  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
[... cut ...]
wcwidth                           0.2.5   c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip
wheel                             0.37.0  c:\users\bens pc\appdata\local\programs\mu editor\python\lib\site-packages                             pip

C:\Users\Bens Pc>"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -I -m mu
qt.svg: QSvgHandler: Image filename is empty
qt.svg: QSvgHandler: Image filename is empty
qt.svg: QSvgHandler: Image filename is empty
qt.svg: QSvgHandler: Image filename is empty
```

carlosperate, 2021-12-27:
That is fantastic news, thank you for all the testing @electric1236, it's been invaluable!

I think might be able edit the Mu shortcut to use the `-I` flag as well, that could be a workaround until we fix it in the next release.

electric1236, 2021-12-27:
Thank you for all your help

electric1236, 2022-05-24:
Hi @carlosperate, sorry for the delay; I'll try to test it this weekend, I don't have access to my desktop right now, and I updated my laptop to window 11, which fixed the problem on it. If I don't get back to you by this weekend please remind me. thanks

electric1236, 2022-05-29:
Hi, @carlosperate I tried the installer and it worked! Thank you