mu-editor/mu #1604: Crash Report Failed to import: pgzero

mitag, 2021-05-28:
[mu.log](https://github.com/mu-editor/mu/files/6562025/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: de_DE
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\mike\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\mike\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\mike\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\mike\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\mike\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-05-31:
Thanks for the report and the log @mitag!

I believe it might be this issue: https://github.com/mu-editor/mu/issues/1421#issuecomment-823662098

Could you try uninstalling the Mu editor and then installing again? Hopefully this way it removes any files from the previous installation and should work.

mitag, 2021-06-01:
I tried different things, installed and uninstalled versions 1.1.0b2,1.1.0b3 and 1.1.0b4 several times and ended up with version 1.1.0-alpha.2. In between, I also uninstalled my Python 2.7 and Python 3.9.1 and installed Python 3.9.5 (which should not affect the Mu installation). So, at your request I have now reinstalled version 1.1.0b4, but following a hunch I went to the Windows Control Panel Programs and there were versions 1.1.0b2 and 1.1.0-alpha.2 installed in parallel. The beta versions always recognise and uninstall their predecessors before they start installing, but they probably leave the alpha version untouched. Since I originally switched from version 1.1.0-alpha.2 to 1.1.0b4, I suspect that this is where my problem came from. With 1.1.0b4 as the only version, it now starts without problems, thank you!

closed 2021-06-01 by mitag

carlosperate, 2021-06-01:
I'm glad you were able to resolve it, thank you for sharing the steps you took @mitag!