mu-editor/mu #1606: i2c write byte array

gmc-code, 2021-05-30:
# Bug Report

What were you trying to do?
---------------------------
maqueen bot driving motor with 3 bytes.
i2c.write(i2c_addr, buf, repeat=False)  where buf = bytearray(3)

What steps did you take to trigger the issue?
---------------------------------------------
using i2c to drive motor

What did you expect to happen?
------------------------------
motor to work

What actually happened?
-----------------------
OS error 19

Why is this difference problematic? (it might not be a bug!)
----------------------------------------------------------
Works fine with Thonny

Technical Info
--------------
* Mu version: 1.1.0 beta 4 and 1.0.3
* Operating system version: windows 10

Other Info
----------

Code that works on Thonny:

from microbit import *

i2c.init(freq=100000, sda=pin20, scl=pin19) # default
# i2c.init()
i2c_addr = 0x10

# def run(mot=0, sens=1, vit=25):
def run(mot, sens, vit):
    buf = bytearray(3)
    if mot == 0:
        buf[0] = 0x00
    else:
        buf[0] = 0x02
    buf[1] = sens
    buf[2] = vit
    i2c.write(i2c_addr, buf, repeat=False)

run(1, 0, 0)
run(0, 0, 0)

while True:
    run(1, 0, 25)
    run(0, 0, 25)
    sleep(500)

Editor Log
----------

<details> <summary>Mu Log Files</summary>
  <pre>
2021-05-30 11:38:31,035 - mu.logic:789(_load) DEBUG: from microbit import *

i2c.init(freq=100000, sda=pin20, scl=pin19) # default
# i2c.init()
i2c_addr = 0x10

# def run(mot=0, sens=1, vit=25):
[... 29 lines cut ...]
2021-05-30 11:38:35,087 - mu.modes.microbit:328(flash) INFO: Checking target device.
2021-05-30 11:38:35,589 - mu.modes.microbit:332(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '2.0.0-beta.4', 'version': 'micro:bit v2.0.0-beta.4+7c8e3e8 on 2021-02-19; MicroPython v1.13 on 2021-02-19', 'machine': 'micro:bit with nRF52833'}
2021-05-30 11:38:35,589 - mu.modes.microbit:351(flash) INFO: Board MicroPython: 2.0.0-beta.4
2021-05-30 11:38:35,589 - mu.modes.microbit:353(flash) INFO: Mu MicroPython: 1.0.1
2021-05-30 11:38:35,589 - mu.modes.microbit:505(copy_main) INFO: Copying main.py onto device
2021-05-30 11:38:35,589 - mu.modes.microbit:515(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'from microbit import *\\n\\ni2c.init(freq=100000, sda=pin20, scl=pin')", "f(b'19) # default\\n# i2c.init()\\ni2c_addr = 0x10\\n\\n# def run(mot=0, sen')", "f(b's=1, vit=25):\\ndef run(mot, sens, vit):\\n    buf = bytearray(3)\\n  ')", "f(b'  if mot == 0:\\n        buf[0] = 0x00\\n    else:\\n        buf[0] = ')", "f(b'0x02\\n    buf[1] = sens\\n    buf[2] = vit\\n    i2c.write(i2c_addr, ')", "f(b'buf, repeat=False)\\n\\nrun(1, 0, 0)\\nrun(0, 0, 0)\\n\\nwhile True:\\n    r')", "f(b'un(1, 0, 25)\\n    run(0, 0, 25)\\n    sleep(500)')", 'fd.close()']
2021-05-30 11:38:36,116 - mu.modes.microbit:518(copy_main) INFO: (b'', b'')
2021-05-30 11:38:43,441 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
2021-05-30 11:38:43,441 - mu.modes.microbit:249(flash) DEBUG: Python script:
2021-05-30 11:38:43,441 - mu.modes.microbit:250(flash) DEBUG: b'from microbit import *\n\ni2c.init(freq=100000, sda=pin20, scl=pin19) # default\n# i2c.init()\ni2c_addr = 0x10\n\n# def run(mot=0, sens=1, vit=25):\ndef run(mot, sens, vit):\n    buf = bytearray(3)\n    if mot == 0:\n        buf[0] = 0x00\n    else:\n        buf[0] = 0x02\n    buf[1] = sens\n    buf[2] = vit\n    i2c.write(i2c_addr, buf, repeat=False)\n\nrun(1, 0, 0)\nrun(0, 0, 0)\n\nwhile True:\n    run(1, 0, 25)\n    run(0, 0, 25)\n    sleep(500)'
2021-05-30 11:38:43,441 - mu.modes.microbit:297(flash) INFO: Path to micro:bit: D:\
2021-05-30 11:38:43,443 - mu.modes.base:227(find_device) INFO: Found device on port: COM6
2021-05-30 11:38:43,443 - mu.modes.base:228(find_device) INFO: Serial number: 9904360259994E45004A401700000029000000009796990B
2021-05-30 11:38:43,443 - mu.modes.microbit:302(flash) INFO: Serial port: COM6
2021-05-30 11:38:43,443 - mu.modes.microbit:303(flash) INFO: Device serial number: 9904360259994E45004A401700000029000000009796990B
2021-05-30 11:38:43,443 - mu.modes.microbit:328(flash) INFO: Checking target device.
2021-05-30 11:38:43,943 - mu.modes.microbit:332(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '2.0.0-beta.4', 'version': 'micro:bit v2.0.0-beta.4+7c8e3e8 on 2021-02-19; MicroPython v1.13 on 2021-02-19', 'machine': 'micro:bit with nRF52833'}
2021-05-30 11:38:43,943 - mu.modes.microbit:351(flash) INFO: Board MicroPython: 2.0.0-beta.4
2021-05-30 11:38:43,943 - mu.modes.microbit:353(flash) INFO: Mu MicroPython: 1.0.1
2021-05-30 11:38:43,943 - mu.modes.microbit:505(copy_main) INFO: Copying main.py onto device
2021-05-30 11:38:43,943 - mu.modes.microbit:515(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'from microbit import *\\n\\ni2c.init(freq=100000, sda=pin20, scl=pin')", "f(b'19) # default\\n# i2c.init()\\ni2c_addr = 0x10\\n\\n# def run(mot=0, sen')", "f(b's=1, vit=25):\\ndef run(mot, sens, vit):\\n    buf = bytearray(3)\\n  ')", "f(b'  if mot == 0:\\n        buf[0] = 0x00\\n    else:\\n        buf[0] = ')", "f(b'0x02\\n    buf[1] = sens\\n    buf[2] = vit\\n    i2c.write(i2c_addr, ')", "f(b'buf, repeat=False)\\n\\nrun(1, 0, 0)\\nrun(0, 0, 0)\\n\\nwhile True:\\n    r')", "f(b'un(1, 0, 25)\\n    run(0, 0, 25)\\n    sleep(500)')", 'fd.close()']
2021-05-30 11:38:44,471 - mu.modes.microbit:518(copy_main) INFO: (b'', b'')
2021-05-30 11:39:22,519 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\gmccarthy\AppData\Local\python\mu\Logs\mu.log
  </pre>
</details>

carlosperate, 2021-07-09:
Hi @gmc-code, thanks for the report!
The MicroPython version in Mu has been updated, could you try beta.5 to see if it fixes the issue?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

gmc-code, 2021-08-12:
All good now, thanks. Code above works now. Much appreciated.

carlosperate, 2021-08-12:
Glad it is now working and thanks for the update! I'll close this issue as resolved then 👍

closed 2021-08-12 by carlosperate