mu-editor/mu #1607: pip3 install mu-editor==1.1.0b4Crash Report

lennoclaup, 2021-05-30:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l

```
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/pi/.local/lib/python3.7/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/pi/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "/home/pi/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "/home/pi/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 603, in ensure_key_modules
    "Failed to import: %s" % module
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-05-31:
Thanks for the report @lennoclaup!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

Could you also let us know what Operating system are you using? I assume this might be in a Raspberry Pi with Raspberry Pi OS Buster?

carlosperate, 2021-06-01:
Hi @lennoclaup,

We didn't really get any new information from your email response, if you tried to attach a file unfortunately it doesn't work via email, you'll have to drag and drop it into the text editor the GitHub website: https://github.com/mu-editor/mu/issues/1607

carlosperate, 2021-08-11:
Hi @lennoclaup,

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5
`pip install mu-editor==1.1.0b5`

Thanks!

carlosperate, 2021-09-21:
@lennoclaup have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate