mu-editor/mu #1609: Crash Report

RE2222, 2021-05-31:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

RE2222, 2021-05-31:
I just loaded Python and MU to a computer and keep getting this error with MU.  
This worked on another weaker computer without a crash.

carlosperate, 2021-06-14:
Thanks the report @Muthuswamy18!

I think this problem is already captured in issue https://github.com/mu-editor/mu/issues/1532, so to keep things tidy I'll merge this over there: https://github.com/mu-editor/mu/issues/1532#issuecomment-860613434

If you could attach your log file in that issue that would still be quite useful, thanks!

closed 2021-06-14 by carlosperate