mu-editor/mu #1612: Crash Report

efremovru, 2021-06-01:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-234955

```

efremovru, 2021-06-01:
2021-05-31 23:09:30,682 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-31 23:09:30,685 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-5JK3UUO', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')
2021-05-31 23:09:30,685 - root:267(run) INFO: Python path: ['C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-31 23:09:30,686 - root:268(run) INFO: Language code: en_US
2021-05-31 23:09:31,204 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-31 23:09:32,298 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-31 23:09:32,299 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230909
2021-05-31 23:09:32,299 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210531-230909
2021-05-31 23:09:35,100 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'
2021-05-31 23:09:35,101 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230909 as C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230909.FAILED
Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230909

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\rumen\\AppData\\Local\\python\\mu\\mu_venv-38-20210531-230909' -> 'C:\\Users\\rumen\\AppData\\Local\\python\\mu\\mu_venv-38-20210531-230909.FAILED'
2021-05-31 23:09:35,108 - mu.virtual_environment:410(relocate) DEBUG: Virtual environment set up mu_venv-38-20210531-230935 at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230935
2021-05-31 23:09:35,109 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x000002CDD91FD790>; Splash Handler; <SplashLogHandler (INFO)>
2021-05-31 23:09:35,109 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #2.
2021-05-31 23:09:35,109 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230935
2021-05-31 23:09:35,109 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210531-230935
2021-05-31 23:09:35,620 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'
2021-05-31 23:09:35,620 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230935 as C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230935.FAILED
Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
[... cut ...]
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-234955
2021-05-31 23:50:03,099 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\session.json
2021-05-31 23:50:03,101 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\venv.json

carlosperate, 2021-06-02:
Thank you very much for the report and the log @efremovru! 

Two interesting bit, first an unexpected error from the `virtualenv` module and then an issue quarantining the failed venv

### `ModuleNotFoundError` in `virtualenv`

This is the main error disrupting the venv creation:
```
2021-05-31 23:09:32,299 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-230909
2021-05-31 23:09:32,299 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210531-230909
2021-05-31 23:09:35,100 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'
```
This is the first time I'm seeing this specific issue. A quick google search suggest issues with the `virtualenv` module installed from package managers like aptitude in Debian/Ubuntu, but this should be the version packaged by Mu.

@efremovru could you confirm you've install Mu from the beta.4 installer? 

### `FileNotFoundError` quarantining
```
Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\rumen\\AppData\\Local\\python\\mu\\mu_venv-38-20210531-230935' -> 'C:\\Users\\rumen\\AppData\\Local\\python\\mu\\mu_venv-38-20210531-230935.FAILED'
```

This second issue is minor as is just an issue renaming the failed venv. We've seen this one before and been collecting them here: https://github.com/mu-editor/mu/issues/1556

It's possible antivirus might be affecting this Mu operation. @efremovru do you have an antivirus running that you could temporarily deactivate when launching Mu?
I'd expect the first problem with the `virtualenv` module to still be present, but it would be really useful to us if you could try running Mu without the antivirus and send us the logs, to see the second problem is still present that way.

efremovru, 2021-06-02:
Thanks you for your reply. Please see my answers below

 

Yes, positive. 

 

Done (I think 😊) 

The results are attached

carlosperate, 2021-06-02:
Thanks @efremovru!

If you attached the log to an email, then unfortunately those don't get posted to GitHub, you'll have to go to the GitHub website and add it as a response (you can drag&drop the file into the text editor):  https://github.com/mu-editor/mu/issues/1612

efremovru, 2021-06-02:
2021-06-02 13:00:37,180 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-06-02 13:00:37,183 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-5JK3UUO', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')
2021-06-02 13:00:37,183 - root:267(run) INFO: Python path: ['C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-06-02 13:00:37,184 - root:268(run) INFO: Language code: en_US
2021-06-02 13:00:37,897 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-06-02 13:00:38,968 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-06-02 13:00:38,968 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-234955
2021-06-02 13:00:38,987 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210531-234955
2021-06-02 13:00:41,261 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'
2021-06-02 13:00:41,261 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-234955 as C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-234955.FAILED
Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210531-234955

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\rumen\\AppData\\Local\\python\\mu\\mu_venv-38-20210531-234955' -> 'C:\\Users\\rumen\\AppData\\Local\\python\\mu\\mu_venv-38-20210531-234955.FAILED'
2021-06-02 13:00:41,280 - mu.virtual_environment:410(relocate) DEBUG: Virtual environment set up mu_venv-38-20210602-130041 at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130041
2021-06-02 13:00:41,280 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x000001F7045A7D30>; Splash Handler; <SplashLogHandler (INFO)>
2021-06-02 13:00:41,280 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #2.
2021-06-02 13:00:41,281 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130041
2021-06-02 13:00:41,281 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210602-130041
2021-06-02 13:00:41,785 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'
2021-06-02 13:00:41,785 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130041 as C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130041.FAILED
Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
[... cut ...]
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130820
2021-06-02 13:08:28,021 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\session.json
2021-06-02 13:08:28,023 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\venv.json

carlosperate, 2021-06-04:
Thank you for trying again and sending the log @efremovru!
 
Just to confirm, you did have an antivirus running the first time you run Mu, and then the second time (when the new log was sent) you disabled the antivirus before opening Mu and getting the same error again?

efremovru, 2021-06-04:
Yes, that is correct. In my email, I attached a screenshot of MS antivirus being disabled. 

 

The annoying part is that I was able to install the software back in Feb 21 and there were no problems. I uninstalled it because I was not using Adafruit ESP, now I got one so I decided to give it a try again. 

 

Is it possible that back in Feb you had older version of the software?

carlosperate, 2021-06-04:
Thanks for the quick response @efremovru, the email attachments don't get posted in GitHub, so we didn't get any screenshots from that email either.

The last log that you copy/pasted (thanks again!), shows that the second problem still happens even with the antivirus disabled:
```
os.rename(self.path, error_dirpath)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130820' -> 'C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130820.FAILED'
```

In February the way Mu worked was different, so some of the new components are having issues in your environment.

The first problem (`ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'`) could happen if maybe the installation didn't complete successfully. Could you try uninstall the Mu Editor, install it again, and see if that works?
If it doesn't, then a fresh copy of the log would be helpful as well.

efremovru, 2021-06-04:
I will, but trust me I tried this install/uninstall 10 times before I entered the issue

carlosperate, 2021-06-04:
Did you also try downloading the installer again?

efremovru, 2021-06-04:
Nope, but will do it now

carlosperate, 2021-06-04:
If after all that it still doesn't work, could you check if these file exists in the Mu installation?

```
C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\Lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py
```

and

```
C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\Lib\site-packages\virtualenv\seed\embed\via_app_data\__init__.py
```

efremovru, 2021-06-04:
It did not work.

 

1.	Fresh download of the executable
2.	Disable Anti-Virus
3.	Disable VPN
4.	Run the executable
5.	Try to open mu
6.	Error message

 

Yes, both files exist

 

And here is the log

 

 

2021-06-04 09:24:49,673 - root:265(run) INFO: 

 

-----------------

 

Starting Mu 1.1.0.beta.4

2021-06-04 09:24:49,677 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-5JK3UUO', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')

2021-06-04 09:24:49,677 - root:267(run) INFO: Python path: ['C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']

2021-06-04 09:24:49,677 - root:268(run) INFO: Language code: en_US

2021-06-04 09:24:50,247 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.

2021-06-04 09:24:51,120 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.

2021-06-04 09:24:51,120 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130820

2021-06-04 09:24:51,135 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210602-130820

2021-06-04 09:24:53,194 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'

2021-06-04 09:24:53,195 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130820 as C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130820.FAILED

Traceback (most recent call last):

  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create

    self.create()

  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create

    self.create_venv()

  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv

    raise VirtualEnvironmentCreateError(

mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210602-130820

 

During handling of the above exception, another exception occurred:
[... 195 lines cut ...]
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run

    raise ex

  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run

    venv.ensure_and_create(self.display_text)

  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create

    self.create()

  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create

    self.create_venv()

  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv

    raise VirtualEnvironmentCreateError(

mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210604-093131

2021-06-04 09:31:39,488 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\session.json

2021-06-04 09:31:39,489 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\venv.json

carlosperate, 2021-06-04:
Thank you @efremovru.

If the files are present but the Mu logs still say that it cannot find the `virtualenv.seed.embed.via_app_data` module, then I think it's possible maybe there is a Python `virtualenv` package executable in the PATH taking preference.

Do you use Python in your computer, and if so do you know how it is normally invoked (in Windows depending on how Python is installed it might be executed in a terminal with`python` or `py`. Do you have several Python versions installed?

Can you try to run these commands in a command prompt and copy/paste the results here?

```
python -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
``` 
```
python3 -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
``` 
```
py -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
```
```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
```

efremovru, 2021-06-04:
I had python installed but then when I started having problems with mu, I uninstalled it just to see if that will make any difference. It did not. I haven’t re-install it pending the resolution of the mu issue. Should I re-install?

carlosperate, 2021-06-04:
No need to reinstall, but if you could try those 4 commands independently that will be useful.

(Btw, if you are seeing my last responses in your email it might be a bit more readable from the GitHub website:  https://github.com/mu-editor/mu/issues/1612)

carlosperate, 2021-06-04:
Oh, if you could try this command as well it will be useful too:

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"
```

efremovru, 2021-06-04:
Done. Pasted the results below. Do they make sense?

 

python -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
 
python3 -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.
 
py -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
'py' is not recognized as an internal or external command,
operable program or batch file.

 

"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"

['', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']

carlosperate, 2021-06-04:
Thanks @efremovru!

I think you missed this one, which was casually the most important one 😅

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv.seed.embed.via_app_data; print(virtualenv.seed.embed.via_app_data.__file__)"
```

efremovru, 2021-06-04:
Apologies. Here is the result

 

Traceback (most recent call last):

  File "<string>", line 1, in <module>

ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'

carlosperate, 2021-06-04:
Thanks!
Okay, that error is good, means we can replicate by invoking the Python interpretert embedded inside Mu.

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list
```

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv; print(virtualenv.__version__, virtualenv.__file__, dir(virtualenv), sep='\n')"
```

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv; print(dir(virtualenv.seed))"
```

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv; print(dir(virtualenv.seed.embed))"
```

efremovru, 2021-06-04:
Here it is. Apologies for the delay.

 

C:\Users\rumen>"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list

Package                Version

---------------------- --------

adafruit-board-toolkit 1.0.1

appdirs                1.4.3

backcall               0.2.0

black                  20.8b1

click                  7.1.2

colorama               0.4.4

decorator              5.0.7

distlib                0.3.0

filelock               3.0.12

flake8                 3.9.1

ipykernel              5.5.3

ipython                7.22.0

ipython-genutils       0.2.0

jedi                   0.18.0

jupyter-client         6.1.12

jupyter-core           4.7.1

mccabe                 0.6.1

mu-editor              1.1.0b4

mypy-extensions        0.4.3

nudatus                0.0.5

parso                  0.8.2

pathspec               0.8.1

pickleshare            0.7.5

pip                    20.2.2

prompt-toolkit         3.0.18

pycodestyle            2.7.0

pyflakes               2.3.1

Pygments               2.8.1

PyQt5                  5.13.2

PyQt5-sip              12.8.1

PyQtChart              5.13.1

pyserial               3.5

python-dateutil        2.8.1

pywin32                300

pyzmq                  22.0.3

QScintilla             2.11.3

qtconsole              4.7.7

QtPy                   1.9.0

regex                  2021.4.4

semver                 2.13.0

setuptools             49.6.0

six                    1.15.0

toml                   0.10.2

tornado                6.1

traitlets              5.0.5

typed-ast              1.4.3

typing-extensions      3.7.4.3

virtualenv             20.0.18

wcwidth                0.2.5

wheel                  0.36.2

 

C:\Users\rumen>"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv; print(virtualenv.__version__, virtualenv.__file__, dir(virtualenv), sep='\n')"

20.0.18

C:\Users\rumen\AppData\Roaming\Python\Python38\site-packages\virtualenv\__init__.py

['__all__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__version__', 'absolute_import', 'cli_run', 'config', 'create', 'discovery', 'info', 'pyenv_cfg', 'report', 'run', 'session', 'unicode_literals', 'util', 'version']

 

C:\Users\rumen>"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv; print(dir(virtualenv.seed))"

Traceback (most recent call last):

  File "<string>", line 1, in <module>

AttributeError: module 'virtualenv' has no attribute 'seed'

 

C:\Users\rumen>"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import virtualenv; print(dir(virtualenv.seed.embed))"

Traceback (most recent call last):

  File "<string>", line 1, in <module>

AttributeError: module 'virtualenv' has no attribute 'seed'

 

Could you try these commands:

efremovru, 2021-07-02:
guys, any chance i can find an older version of mu somewhere?

carlosperate, 2021-07-05:
Apologies for the delay, we've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

efremovru, 2021-07-05:
thanks Calros. i just tried it and the problem is still there

Traceback (most recent call last):
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210705-181446
ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'

and here is the full log:

2021-07-05 17:44:52,820 - mu.interface.main:751(show_confirmation) DEBUG: There is un-saved work, exiting the application will cause you to lose it.
2021-07-05 17:44:52,846 - mu.interface.main:752(show_confirmation) DEBUG: None
2021-07-05 17:44:54,573 - mu.logic:1053(quit) DEBUG: Session: {'theme': 'night', 'mode': 'adafruit', 'paths': ['D:\\bno055_simpletest.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 1}
2021-07-05 17:44:54,574 - mu.logic:1054(quit) DEBUG: Saving session to: C:\Users\rumen\AppData\Local\python\mu\session.json
2021-07-05 17:44:54,585 - mu.logic:1068(quit) INFO: Quitting.

2021-07-05 18:14:42,344 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.5
2021-07-05 18:14:42,344 - root:272(run) INFO: uname_result(system='Windows', node='DESKTOP-5JK3UUO', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 61 Stepping 4, GenuineIntel')
2021-07-05 18:14:42,345 - root:273(run) INFO: Platform: Windows-10-10.0.19041-SP0
2021-07-05 18:14:42,345 - root:274(run) INFO: Python path: ['C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-07-05 18:14:42,345 - root:275(run) INFO: Language code: en_US
2021-07-05 18:14:42,788 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-07-05 18:14:43,380 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-07-05 18:14:43,380 - mu.virtual_environment:777(create_venv) INFO: Creating virtualenv: C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210703-134255
2021-07-05 18:14:43,398 - mu.virtual_environment:778(create_venv) INFO: Virtualenv name: mu_venv-38-20210703-134255
2021-07-05 18:14:43,401 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\rumen\\AppData\\Local\\python\\mu\\mu_venv-38-20210703-134255') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\rumen\\AppData\\Roaming', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'DESKTOP-5JK3UUO', 'COMSPEC': 'C:\\WINDOWS\\system32\\cmd.exe', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'HOMEDRIVE': 'C:', 'HOMEPATH': '\\Users\\rumen', 'LOCALAPPDATA': 'C:\\Users\\rumen\\AppData\\Local', 'LOGONSERVER': '\\\\DESKTOP-5JK3UUO', 'NUMBER_OF_PROCESSORS': '4', 'ONEDRIVE': 'C:\\Users\\rumen\\OneDrive', 'ONEDRIVECONSUMER': 'C:\\Users\\rumen\\OneDrive', 'OS': 'Windows_NT', 'PATH': 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\Program Files\\AdoptOpenJDK\\jdk-11.0.10.9-hotspot\\bin;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\Git\\cmd;C:\\Users\\rumen\\AppData\\Local\\Programs\\Python\\Python39;C:\\Users\\rumen\\AppData\\Local\\Programs\\Python\\Python39\\Scripts;C:\\Program Files\\PuTTY [... cut ...]
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\rumen\AppData\Local\python\mu\mu_venv-38-20210705-181446
ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'
2021-07-05 18:14:54,654 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\session.json
2021-07-05 18:14:54,656 - mu.settings:169(save) DEBUG: Saving to C:\Users\rumen\AppData\Local\python\mu\venv.json

i went back through the versions and the one which actually works for me is  mu-editor_1.0.3_win64

carlosperate, 2021-08-05:
Thanks for trying again. Looking at your system configuration something that jumped to me was one of the entries to the Python path:

```
2021-07-05 18:14:42,345 - root:274(run) INFO: Python path: [
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python38.zip',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\DLLs',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python',
    'C:\Users\rumen\AppData\Roaming\Python\Python38\site-packages',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\setuptools-49.6.0-py3.8.egg',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\pip-20.2.2-py3.8.egg',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\win32',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\win32\lib',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\Pythonwin',
    'C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\IPython\extensions'
]
```

I did not expect to see `C:\Users\rumen\AppData\Roaming\Python\Python38\site-packages`.

Could you run the following commands in a command prompt and copy/paste the results here?

```
dir C:\Users\rumen\AppData\Roaming\Python\Python38\site-packages
```

```
py -c "import sys; print(sys.path)"
```

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"
```

efremovru, 2021-08-05:
C:\Users\rumen>dir C:\Users\rumen\AppData\Roaming\Python\Python38\site-packages

Volume in drive C has no label.

Volume Serial Number is 5E4F-77FA

 

Directory of C:\Users\rumen\AppData\Roaming\Python\Python38\site-packages

 

04/18/2020  09:28 PM    <DIR>          .

04/18/2020  09:28 PM    <DIR>          ..

04/18/2020  09:27 PM    <DIR>          appdirs-1.4.3.dist-info

04/18/2020  09:27 PM            24,701 appdirs.py

04/18/2020  09:27 PM    <DIR>          distlib

04/18/2020  09:27 PM    <DIR>          distlib-0.3.0-py3.8.egg-info

04/18/2020  09:27 PM    <DIR>          filelock-3.0.12.dist-info

04/18/2020  09:27 PM            13,229 filelock.py

04/18/2020  09:28 PM    <DIR>          virtualenv

04/18/2020  09:28 PM    <DIR>          virtualenv-20.0.18.dist-info

04/18/2020  09:27 PM    <DIR>          __pycache__

               2 File(s)         37,930 bytes

               9 Dir(s)  49,707,954,176 bytes free

 

C:\Users\rumen>py -c "import sys; print(sys.path)" "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"

['', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Python\\Python39\\python39.zip', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Python\\Python39\\DLLs', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Python\\Python39\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Python\\Python39', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python39\\site-packages', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python39\\site-packages\\win32', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python39\\site-packages\\win32\\lib', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python39\\site-packages\\Pythonwin', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Python\\Python39\\lib\\site-packages']

carlosperate, 2021-08-05:
Thanks @efremovru! 

I think two commands got mixed up:

> C:\Users\rumen>py -c "import sys; print(sys.path)" "C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)" ['', [...]

This lists only `C:\Users\rumen>py -c "import sys; print(sys.path)"` could you try this command separately as well?

```
"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"
```

efremovru, 2021-08-05:
Carlos, apologies, because of the quote marks I thought that the last line was part of the command above. Here is the correct version

 

 

Microsoft Windows [Version 10.0.19043.1110]

(c) Microsoft Corporation. All rights reserved.

 

C:\Users\rumen>"C:\Users\rumen\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"

['', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\rumen\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\rumen\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']

 

C:\Users\rumen>

carlosperate, 2021-12-15:
@efremovru  We think we might have found a solution to this problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.

efremovru, 2021-12-15:
Not sure how to do it. I thought that there will be a link behind the name but they seem to be static labels 

 

https://github.com/mu-editor/mu/actions/runs/1574840282

carlosperate, 2021-12-15:
Right, make sure you are logged into GitHub when going to https://github.com/mu-editor/mu/actions/runs/1574840282 , otherwise it won't add the link to the downloads.

efremovru, 2021-12-17:
Houston, we have lift off!!!!!!!!!!!!!!!!!

 

Thank you Carlos! All good now. The universe is at piece😊

carlosperate, 2021-12-17:
Awesome, thanks for trying the installer!
I'll close this issue as we are tracking the problem and fix via https://github.com/mu-editor/mu/issues/1831.

Thanks again for bearing with us this long and helping with all the debugging!

closed 2021-12-17 by carlosperate