mu-editor/mu #1613: Every time I open Mu Editor in my Raspberry Pi the window gets higher

aleggon, 2021-06-01:
# Bug Report

What were you trying to do?
---------------------------
Open the editor

What steps did you take to trigger the issue?
---------------------------------------------
Just installed via pip

What did you expect to happen?
------------------------------
I expected to get the window always at the same size and position on the escreen

What actually happened?
-----------------------
Every time I open the editor the window opens at a higher position in the screen

Why is this difference problematic? (it might not be a bug!)
----------------------------------------------------------
Well, sometimes I just can't reach the top of the window to reposition it and so that I can't use the app.

Technical Info
--------------
* Mu version: Mu 1.1.0.beta.4
* Operating system version: Raspberry Pi OS (kernel version: 5.10)

Other Info
----------

Editor Log
----------

<div class="adn ads" style="display:" data-message-id="#msg-a:r5545924778712747979" data-legacy-message-id="179c7173abda0030"><div class="gs"><div class=""><div id=":pd" class="ii gt"><div id=":pc" class="a3s aiL "><div dir="ltr"><p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:09,848 - root:265(run) INFO: </p>
<p style="margin:0px;white-space:pre-wrap"><br></p>
<p style="margin:0px;white-space:pre-wrap">-----------------</p>
<p style="margin:0px;white-space:pre-wrap"><br></p>
<p style="margin:0px;white-space:pre-wrap">Starting Mu 1.1.0.beta.4</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:09,849 - root:266(run) INFO: uname_result(system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1421 SMP Thu May 27 14:00:13 BST 2021', machine='armv7l', processor='')</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:09,849 - root:267(run) INFO: Python path: ['/home/pi/.local/bin', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-<wbr>dynload', '/home/pi/.local/lib/python3.<wbr>7/site-packages', '/usr/local/lib/python3.7/<wbr>dist-packages', '/usr/lib/python3/dist-<wbr>packages', '/usr/lib/python3/dist-<wbr>packages/IPython/extensions']</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:09,850 - root:268(run) INFO: Language code: pt_BR</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,601 - mu.virtual_environment:480(<wbr>ensure_and_create) INFO: Added log handler.</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,668 - mu.virtual_environment:491(<wbr>ensure_and_create) DEBUG: Checking virtual environment; attempt #1.</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,669 - mu.virtual_environment:542(<wbr>ensure_path) INFO: Virtual Environment found at: /home/pi/.local/share/mu/mu_<wbr>venv-37-20210530-182456</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,686 - mu.virtual_environment:555(<wbr>ensure_interpreter) INFO: Interpreter found at: /home/pi/.local/share/mu/mu_<wbr>venv-37-20210530-182456/bin/<wbr>python</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,776 - mu.virtual_environment:382(<wbr>run_subprocess) DEBUG: Process returned 0; output: 37</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,777 - mu.virtual_environment:585(<wbr>ensure_interpreter_version) INFO: Both interpreters at version 37</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,778 - mu.virtual_environment:611(<wbr>ensure_pip) INFO: Pip found at: /home/pi/.local/share/mu/mu_<wbr>venv-37-20210530-182456/bin/<wbr>pip</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,799 - mu.virtual_environment:597(<wbr>ensure_key_modules) DEBUG: Verifying import of: pgzero</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,884 - mu.virtual_environment:382(<wbr>run_subprocess) DEBUG: Process returned 0; output: </p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:10,885 - mu.virtual_environment:597(<wbr>ensure_key_modules) DEBUG: Verifying import of: flask</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,613 - mu.virtual_environment:382(<wbr>run_subprocess) DEBUG: Process returned 0; output: </p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,614 - mu.virtual_environment:597(<wbr>ensure_key_modules) DEBUG: Verifying import of: qtconsole</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,700 - mu.virtual_environment:382(<wbr>run_subprocess) DEBUG: Process returned 0; output: </p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,701 - mu.virtual_environment:597(<wbr>ensure_key_modules) DEBUG: Verifying import of: esptool</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,892 - mu.virtual_environment:382(<wbr>run_subprocess) DEBUG: Process returned 0; output: </p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,893 - mu.virtual_environment:597(<wbr>ensure_key_modules) DEBUG: Verifying import of: adafruit_board_toolkit</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,981 - mu.virtual_environment:382(<wbr>run_subprocess) DEBUG: Process returned 0; output: </p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,982 - mu.virtual_environment:496(<wbr>ensure_and_create) INFO: Valid virtual environment found at /home/pi/.local/share/mu/mu_<wbr>venv-37-20210530-182456</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,983 - mu.settings:169(save) DEBUG: Saving to /home/pi/.local/share/mu/venv.<wbr>json</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:11,985 - mu.virtual_environment:511(<wbr>ensure_and_create) DEBUG: Emitter: &lt;bound PYQT_SIGNAL display_text of StartupWorker object at 0xb0566cb0&gt;; Splash Handler; &lt;SplashLogHandler (INFO)&gt;</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,012 - mu.logic:758(__init__) INFO: Setting up editor.</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,013 - mu.logic:778(__init__) INFO: Log directory: /home/pi/.cache/mu/log</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,014 - mu.logic:779(__init__) INFO: Data directory: /home/pi/.local/share/mu</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,015 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,016 - mu.modes.base:63(get_default_<wbr>workspace) INFO: Using workspace /home/pi/mu_code from settings file</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,141 - mu.modes.base:63(get_default_<wbr>workspace) INFO: Using workspace /home/pi/mu_code from settings file</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,141 - mu.logic:1521(change_mode) INFO: Workspace directory: /home/pi/mu_code</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,143 - mu.logic:867(restore_session) DEBUG: &lt;SessionSettings from /home/pi/.local/share/mu/<wbr>session.json&gt;</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:44:12,167 - mu.logic:889(restore_session) INFO: Loaded files.</p>
[... 625 lines cut ...]
<p style="margin:0px;white-space:pre-wrap"><br></p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:56:32,056 - mu.settings:169(save) DEBUG: Saving to /home/pi/.local/share/mu/<wbr>session.json</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:56:32,057 - mu.settings:169(save) DEBUG: Saving to /home/pi/.local/share/mu/venv.<wbr>json</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:56:40,697 - root:265(run) INFO: </p>
<p style="margin:0px;white-space:pre-wrap"><br></p>
<p style="margin:0px;white-space:pre-wrap">-----------------</p>
<p style="margin:0px;white-space:pre-wrap"><br></p>
<p style="margin:0px;white-space:pre-wrap">Starting Mu 1.1.0.beta.4</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:56:40,698 - root:266(run) INFO: uname_result(system='Linux', node='raspberrypi', release='5.10.17-v7l+', version='#1421 SMP Thu May 27 14:00:13 BST 2021', machine='armv7l', processor='')</p>
<p style="margin:0px;white-space:pre-wrap">2021-06-01 06:56:40,698 - root:267(run) INFO: Python path: ['/home/pi/.local/bin', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-<wbr>dynload', '/home/pi/.local/lib/python3.<wbr>7/site-packages', '/usr/local/lib/python3.7/<wbr>dist-packages', '/usr/lib/python3/dist-<wbr>packages', '/usr/lib/python3/di [... cut ...]
</div></div></div></div></div></div></div></div></div></div></div><div class="nH nn" style="width: 56px; min-width: 56px;" jsaction="CyXzrf:.CLIENT"><div class="no"><div class="nH bAw nn" style="width: 56px; height: 1794px;"><div class="brC-aT5-aOt-Jw" role="complementary" aria-label="Painel lateral"><div class="brC-aT5-aOt-bsf-Jw"></div></div></div></div></div><div class="brC-aT5-aOt-Jw" role="complementary" aria-label="Painel lateral"></div><div class="brC-dA-I-Jw brC-dA-I-Jw-qk"></div><div class="no"><div class="nH nn" style="width: 56px; min-width: 56px;" jsaction="CyXzrf:.CLIENT"><div class="dJ"></div></div></div><div class="dJ"></div>

carlosperate, 2021-06-01:
Weird, I wonder if the Window position saved by Mu in the session file and then used to set the window on the startup might not always be taking in consideration the title bar?

@ntoll do you experience anything similar in Debian? I guess you are probably using Gnome 3, is not that close to the LXDE+gtk2 from RpiOS, but maybe it's a similar enough environment?

carlosperate, 2021-06-02:
I was able to replicate in the x86 Pi OS desktop image.

First placing the window at a height that is easy to check (aligning with the terminal window behind it)
[image]

The next run it's one titlebar higher:
[image]

And a few runs after we go off-screen and we can't move the window around with the mouse:
[image]

@aleggon in the meantime you can manually remove the window position from the json file in `/home/pi/.local/share/mu/session.json` (you can delete the entire file if you don't care too much about preserving other settings, like previously opened files).

Is anybody else able to replicate in any other Linux desktop? @ntoll @ZanderBrown?

ntoll, 2021-07-11:
I cannot reproduce this on Linux with Gnome3. My guess is that it's a bug in the window manager or the PiOS..?

This is a weird one. I wonder if there are similar problems with other Qt5 based apps (SonicPi for instance)..?

PR #1768 (2021-09-20): Ensure Mu always appears on the screen when calculating the window size and position.
Fixes #1613.

I have not been able to test this on PiOS (I don't have a Pi to hand). Can someone (@carlosperate?) who does, please check this works as I think it will. Thank you.
closed 2021-09-20 by ntoll