mu-editor/mu #1614: Crash Report

Timur-ccnp, 2021-06-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 4.15.0-143-generic #147-Ubuntu SMP Wed Apr 14 16:10:11 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/student/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/home/student/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 138, in download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/student/.local/lib/python3.6/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/student/.local/lib/python3.6/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/student/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/student/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/home/student/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download esptool==3.*

```

carlosperate, 2021-06-02:
Thanks for the report @Timur-ccnp!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

carlosperate, 2021-08-12:
Hi @Timur-ccnp,

Are you able to attach the log? File should be in `~/.cache/mu/`.

Would you be able to check if beta.5 has resolved the issue let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@Timur-ccnp have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate