mu-editor/mu #1616: C:\Users\Newsaver\AppData\Local\python\mu\Logs\mu.log

Ocho8a, 2021-06-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: es_SV
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Newsaver\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 283, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Newsaver\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Newsaver\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Newsaver\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Newsaver\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 626, in create
    self.register_baseline_packages()
  File "C:\Users\Newsaver\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 751, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\Newsaver\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 289, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

carlosperate, 2021-06-03:
Thanks for the report @Ocho8a!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging.
I think it's possible you might have dragged the file to the title text field, if that's the case you'd have to drag&drop it into the text field of a new comment here.

File should be located in: `C:\Users\Newsaver\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-08-12:
Hi @Ocho8a,

Are you able to attach the log? File should be in `C:\Users\Newsaver\AppData\Local\python\mu\Logs\mu.log`.

Would you be able to check if beta.5 has resolved the issue let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@BarryButler have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

If you could also attach the log (located in `C:\Users\Newsaver\AppData\Local\python\mu\Logs\mu.log`) that would really help us investigate this issue.

BarryButler, 2021-09-25:
Beta 5 works fine.  Thankyou.

Regards,

Barry Butler
Teacher (Digital Technology, Aviation, Maths, Science)

[2019 Email Signature Block]
------------------------

This message (including attachments) is intended for the addressee named above. It may also be confidential, privileged and/or subject to copyright. If you wish to forward this message to others, you must first obtain the permission of the author. If you are not the addressee named above, you must not disseminate, copy, communicate or otherwise use or take any action in reliance on this message. You understand that any privilege or confidentiality attached to this message is not waived, lost or destroyed because you have received this message in error. If you have received this message in error please notify the sender and delete from any computer. Unless explicitly attributed, the opinions expressed in this message do not necessarily represent the official position or opinions of the State of Queensland or the Department of Education, Training and Employment. Whilst all care has been taken, the Department of Education, Training and Employment disclaims all liability for loss or damage to person or property arising from this message being infected by computer virus or other contamination.

IMPORTANT: This email and any attachments may contain legally privileged, confidential or private information, and may be protected by copyright. You may only use or disclose this information if you are the intended recipient(s) and if you use it in an authorised way. No other person is allowed to use, review, alter, transmit, disclose, distribute, print or copy this email and any attachments without appropriate authorisation.

If you are not the intended recipient(s) and the email was sent to you by mistake, please notify the sender immediately by return email or phone, destroy any hardcopies of this email and any attachments and delete it from your system. Any legal privilege and confidentiality attached to this email is not waived or destroyed by that mistake.

The Department of Education carries out monitoring, scanning and blocking of emails and attachments sent from or to addresses within the Department of Education for the purposes of operating, protecting, maintaining and ensuring appropriate use of its computer network. It is your responsibility to ensure that this email does not contain and is not affected by computer viruses, defects or interference by third parties or replication problems (including incompatibility with your computer system).

The Department of Education does not accept any responsibility for any loss or damage that may result from reliance on, or the use of, any information contained in the email and any attachments.

***************************************************************************************************

carlosperate, 2021-09-26:
Awesome, thanks for the update!

closed 2021-09-26 by carlosperate