mu-editor/mu #1618: Crash Report

walter2505, 2021-06-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: de_AT
Platform: Linux 5.4.0-73-generic #82~18.04.1-Ubuntu SMP Fri Apr 16 15:10:02 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/usr/local/lib/python3.6/dist-packages/mu/wheels/__init__.py", line 138, in download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/mu/app.py", line 161, in run
    raise ex
  File "/usr/local/lib/python3.6/dist-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.6/dist-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1

```

walter2505, 2021-06-02:
home/walpet/.cache/mu/log/mu.log

carlosperate, 2021-06-03:
Hi @walter2505, thanks for the report!

Will you be able to attach the log into a new comment here?
You should be able to drag and drop it into the text box, otherwise you could copy/paste the content.
Thanks!

carlosperate, 2021-09-21:
@walter2505 have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

If you could also attach the log (located in `/home/walpet/.cache/mu/log/mu.log`) that would really help us investigate this issue.

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate