mu-editor/mu #1619: Crash Report

webbbn, 2021-06-04:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\webbb\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\webbb\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\webbb\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\webbb\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\webbb\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

2021-06-03 22:13:09,762 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-06-03 22:13:09,763 - root:266(run) INFO: uname_result(system='Windows', node='Galaxy', release='10', version='10.0.19041', machine='AMD64', processor='AMD64 Family 25 Model 33 Stepping 0, AuthenticAMD')
2021-06-03 22:13:09,763 - root:267(run) INFO: Python path: ['C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Program Files\\PothosSDR\\lib\\python3.8\\site-packages', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\webbb\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\webbb\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-06-03 22:13:09,763 - root:268(run) INFO: Language code: en_US
2021-06-03 22:13:09,763 - mu.settings:220(load) WARNING: No settings file found at C:\Users\webbb\AppData\Local\python\mu\settings.json; skipping
2021-06-03 22:13:09,808 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-06-03 22:13:09,911 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-06-03 22:13:09,912 - mu.virtual_environment:542(ensure_path) INFO: Virtual Environment found at: C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210405-115854
2021-06-03 22:13:09,918 - mu.virtual_environment:555(ensure_interpreter) INFO: Interpreter found at: C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210405-115854\scripts\python.exe
2021-06-03 22:13:10,019 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 38
2021-06-03 22:13:10,019 - mu.virtual_environment:585(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-06-03 22:13:10,020 - mu.virtual_environment:611(ensure_pip) INFO: Pip found at: C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210405-115854\scripts\pip.exe
2021-06-03 22:13:10,021 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-06-03 22:13:10,048 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-03 22:13:10,048 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: flask
2021-06-03 22:13:10,399 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-03 22:13:10,399 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-06-03 22:13:10,427 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-03 22:13:10,427 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-06-03 22:13:10,488 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-03 22:13:10,488 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: adafruit_board_toolkit
2021-06-03 22:13:10,514 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'adafruit_board_toolkit'
2021-06-03 22:13:10,516 - mu.virtual_environment:468(quarantine_failed_venv) INFO: Quarantined C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210405-115854 as C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210405-115854.FAILED
2021-06-03 22:13:10,517 - mu.virtual_environment:410(relocate) DEBUG: Virtual environment set up mu_venv-38-20210603-221310 at C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310
2021-06-03 22:13:10,517 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x0000013C469CD8B0>; Splash Handler; <SplashLogHandler (INFO)>
2021-06-03 22:13:10,517 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #2.
2021-06-03 22:13:10,517 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310
2021-06-03 22:13:10,517 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210603-221310
2021-06-03 22:13:12,433 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-03 22:13:12,433 - mu.virtual_environment:648(create_venv) INFO: Created virtual environment using C:\Users\webbb\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310
2021-06-03 22:13:12,433 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310\scripts\python.exe exists
2021-06-03 22:13:16,977 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: Requirement already satisfied: pip in c:\users\webbb\appdata\local\python\mu\mu_venv-38-20210603-221310\lib\site-packages (21.0.1)
Collecting pip
  Downloading pip-21.1.2-py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.0.1
    Uninstalling pip-21.0.1:
      Successfully uninstalled pip-21.0.1
Successfully installed pip-21.1.2
2021-06-03 22:13:16,978 - mu.virtual_environment:669(upgrade_pip) INFO: Upgraded pip
2021-06-03 22:13:16,978 - mu.virtual_environment:711(install_baseline_packages) INFO: Installing baseline packages.
[... 129 lines cut ...]
2021-06-03 22:13:18,990 - mu.virtual_environment:555(ensure_interpreter) INFO: Interpreter found at: C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310\scripts\python.exe
2021-06-03 22:13:19,015 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 38
2021-06-03 22:13:19,015 - mu.virtual_environment:585(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-06-03 22:13:19,015 - mu.virtual_environment:611(ensure_pip) INFO: Pip found at: C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310\scripts\pip.exe
2021-06-03 22:13:19,015 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-06-03 22:13:19,039 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 1; output: Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pgzero'
2021-06-03 22:13:19,041 - mu.virtual_environment:468(quarantine_failed_venv) INFO: Quarantined C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310 as C:\Users\webbb\AppData\Local\python\mu\mu_venv-38-20210603-221310.FAILED
[... cut ...]
    self.ensure_key_modules()
  File "C:\Users\webbb\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
2021-06-03 22:13:26,159 - mu.settings:169(save) DEBUG: Saving to C:\Users\webbb\AppData\Local\python\mu\session.json
2021-06-03 22:13:26,160 - mu.settings:169(save) DEBUG: Saving to C:\Users\webbb\AppData\Local\python\mu\venv.json

carlosperate, 2021-06-04:
Thank you for the report @webbbn!
I believe the issue you are seeing might be this: https://github.com/mu-editor/mu/issues/1421#issuecomment-823662098

Did you have beta.2 or beta.3 installed after trying beta.4?
Could you try uninstalling the Mu Editor and installing again? Hopefully that should clean up the installation directory and work after that.

webbbn, 2021-06-04:
Yes. I didn't know that I did have beta 3 installed, but that appears to have been the problem. I removed both versions and re-installed beta 4, and it is working now.

Thank you!

carlosperate, 2021-06-04:
Glad you were able to resolve it, and thanks for taking the time to report this issue and test Mu!

closed 2021-06-04 by carlosperate