mu-editor/mu #1620: Installation Error

streagob, 2021-06-05:
REMEMBER TO ATTACH YOUR LOG FILE

Locale: en_FR
Platform: ArchLinux 

commands performed:
```
python3 -m venv testenvformueditor
source testenvformueditor/bin/activate.fish
pip3 install mu-editor
```

[log_mueditor.txt](https://github.com/mu-editor/mu/files/6603050/log_mueditor.txt)

carlosperate, 2021-06-07:
Hi @streagob,

We've tagged the latest beta as "pre-release" in PyPI, so you'll have to either use `--pre` flag or the specific version.

Can you try installing with `pip install mu-editor==1.1.0b4` and see if you are still getting errors?
Btw, if you are familiar with virtual environments I'd highly recommend installing Mu in one.

carlosperate, 2021-07-05:
Yeah, it's expected that older version of Mu will have issues with relatively new versions of Python.

That's a bit odd, we just released beta.5 could you try this? (make sure you use the right version of pip depending on your environment):

```
pip install mu-editor==1.1.0b5 -v
```

carlosperate, 2021-07-05:
Ahh, I just realised you might be using Python 3.9 (or a 3.10 beta), is that the case?
Due to some dependencies Mu at the moment is only compatible with Python version 3.5 to 3.8. If you can install an older version of Python it should work with Mu.

carlosperate, 2021-08-17:
As this was attempted to run in Python 3.9 and that work is tracked in https://github.com/mu-editor/mu/issues/1754 I'll close this to keep things tidy.

Thanks for the report!

closed 2021-08-17 by carlosperate

rezapci, 2023-01-22:
i have the same problem on ubuntu 22.0.4.1 lts
pip _v 22.0.2 with python3 3.10.6 and python 2.7.18

ERROR: Could not find a version that satisfies the requirement mu-editor (from versions: 1.0.0b12, 1.0.0b13, 1.0.0b14, 1.0.0b15, 1.0.0b16, 1.0.0b17, 1.0.0rc1, 1.0.0, 1.0.1, 1.0.2, 1.0.3)
ERROR: No matching distribution found for mu-editor