mu-editor/mu #1621: Crash Report

mis2n, 2021-06-05:
[mu.log](https://github.com/mu-editor/mu/files/6603157/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\matth\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\matth\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\matth\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\matth\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\matth\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\matth\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\matth\AppData\Local\python\mu\mu_venv-38-20210605-145755

```

carlosperate, 2021-07-05:
Thanks for the report and the logs @mis2n! 

We've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-08-12:
Hi @mis2n,

I was wondering if you were able to install the latest beta and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate