mu-editor/mu #1626: OSError: [WinError 87] The parameter is incorrect

mklak, 2021-06-09:
All what I can see is inside the log. After installation, I cannot open Mu.

[mu.log](https://github.com/mu-editor/mu/files/6621724/mu.log)

carlosperate, 2021-06-09:
Thank you for the report and the log @mklak!

This is the important trace:

```
2021-06-09 08:34:29,605 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\Michal\AppData\Local\python\mu\mu_venv-38-20210609-083342\scripts\python.exe exists
2021-06-09 08:34:58,281 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x0000000006887790>; Splash Handler; <SplashLogHandler (INFO)>
2021-06-09 08:35:05,332 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in upgrade_pip
    ok, output = self.run_subprocess(
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 373, in run_subprocess
    process = subprocess.run(
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Michal\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] The parameter is incorrect
```

@mklak are you using Windows 7?
From the log we get this info, but I wanted to confirm:
```
2021-06-09 08:32:09,674 - root:266(run) INFO: uname_result(system='Windows', node='lenovoT410', release='7', version='6.1.7601', machine='AMD64', processor='Intel64 Family 6 Model 37 Stepping 2, GenuineIntel')
```

@tjguk have you seen a similar error to this in the past? I wonder if Windows 7 might be a factor or just a red herring.
A quick google search and I wonder if it might be this: https://stackoverflow.com/a/32382067/775259

carlosperate, 2021-08-12:
Hi @mklak,

Can you try again with beta.5 and if it still fails post the new log? We get more detailed information in the new version. Hopefully this way we'll see what command exactly we are trying to spawn.

https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-12-17:
Hi @mklak, I think this crash might be related to issue: #1831

We think we might have found a solution to that problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.