mu-editor/mu #1629: Crash Report

cbrown56, 2021-06-10:
https://codewith.mu/crash/?v=1.1.0.beta.4&l=en_US&p=RGFyd2luIDE5LjYuMCBEYXJ3aW4gS2VybmVsIFZlcnNpb24gMTkuNi4wOiBUaHUgTWF5ICA2IDAwOjQ4OjM5IFBEVCAyMDIxOyByb290OnhudS02MTUzLjE0MS4zM34xL1JFTEVBU0VfWDg2XzY0IHg4Nl82NA%3D%3D&f=L1VzZXJzL2NocmlzdGluYWJyb3duL0xpYnJhcnkvTG9ncy9tdS9tdS5sb2c%3D&e=VHJhY2ViYWNrIChtb3N0IHJlY2VudCBjYWxsIGxhc3QpOgogIEZpbGUgIi9wcml2YXRlL3Zhci9mb2xkZXJzLzV5L2c0NWx3eTA1M3NiZndtbWtyOXN3Mzl3YzAwMDBnbi9UL0FwcFRyYW5zbG9jYXRpb24vOUZCOUZEQTUtQUEwNy00OEJELThBMTktQTEyNUUxNDg1RDMzL2QvTXUgRWRpdG9yLmFwcC9Db250ZW50cy9SZXNvdXJjZXMvUHl0aG9uL2xpYi9weXRob24zLjgvc2l0ZS1wYWNrYWdlcy9tdS9hcHAucHkiLCBsaW5lIDE2MSwgaW4gcnVuCiAgICByYWlzZSBleAogIEZpbGUgIi9wcml2YXRlL3Zhci9mb2xkZXJzLzV5L2c0NWx3eTA1M3NiZndtbWtyOXN3Mzl3YzAwMDBnbi9UL0FwcFRyYW5zbG9jYXRpb24vOUZCOUZEQTUtQUEwNy00OEJELThBMTktQTEyNUUxNDg1RDMzL2QvTXUgRWRpdG9yLmFwcC9Db250ZW50cy9SZXNvdXJjZXMvUHl0aG9uL2xpYi9weXRob24zLjgvc2l0ZS1wYWNrYWdlcy9tdS9hcHAucHkiLCBsaW5lIDE0OCwgaW4gcnVuCiAgICB2ZW52LmVuc3VyZV9hbmRfY3JlYXRlKHNlbGYuZGlzcGxheV90ZXh0KQogIEZpbGUgIi9wcml2YXRlL3Zhci9mb2xkZXJzLzV5L2c0NWx3eTA1M3NiZndtbWtyOXN3Mzl3YzAwMDBnbi9UL0FwcFRyYW5zbG9jYXRpb24vOUZCOUZEQTUtQUEwNy00OEJELThBMTktQTEyNUUxNDg1RDMzL2QvTXUgRWRpdG9yLmFwcC9Db250ZW50cy9SZXNvdXJjZXMvUHl0aG9uL2xpYi9weXRob24zLjgvc2l0ZS1wYWNrYWdlcy9tdS92aXJ0dWFsX2Vudmlyb25tZW50LnB5IiwgbGluZSA0OTQsIGluIGVuc3VyZV9hbmRfY3JlYXRlCiAgICBzZWxmLmNyZWF0ZSgpCiAgRmlsZSAiL3ByaXZhdGUvdmFyL2ZvbGRlcnMvNXkvZzQ1bHd5MDUzc2Jmd21ta3I5c3czOXdjMDAwMGduL1QvQXBwVHJhbnNsb2NhdGlvbi85RkI5RkRBNS1BQTA3LTQ4QkQtOEExOS1BMTI1RTE0ODVEMzMvZC9NdSBFZGl0b3IuYXBwL0NvbnRlbnRzL1Jlc291cmNlcy9QeXRob24vbGliL3B5dGhvbjMuOC9zaXRlLXBhY2thZ2VzL211L3ZpcnR1YWxfZW52aXJvbm1lbnQucHkiLCBsaW5lIDYyNCwgaW4gY3JlYXRlCiAgICBzZWxmLnVwZ3JhZGVfcGlwKCkKICBGaWxlICIvcHJpdmF0ZS92YXIvZm9sZGVycy81eS9nNDVsd3kwNTNzYmZ3bW1rcjlzdzM5d2MwMDAwZ24vVC9BcHBUcmFuc2xvY2F0aW9uLzlGQjlGREE1LUFBMDctNDhCRC04QTE5LUExMjVFMTQ4NUQzMy9kL011IEVkaXRvci5hcHAvQ29udGVudHMvUmVzb3VyY2VzL1B5dGhvbi9saWIvcHl0aG9uMy44L3NpdGUtcGFja2FnZXMvbXUvdmlydHVhbF9lbnZpcm9ubWVudC5weSIsIGxpbmUgNjcxLCBpbiB1cGdyYWRlX3BpcAogICAgcmFpc2UgVmlydHVhbEVudmlyb25tZW50Q3JlYXRlRXJyb3IoIlVuYWJsZSB0byB1cGdyYWRlIHBpcCIpCm11LnZpcnR1YWxfZW52aXJvbm1lbnQuVmlydHVhbEVudmlyb25tZW50Q3JlYXRlRXJyb3I6IFVuYWJsZSB0byB1cGdyYWRlIHBpcAo%3D
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Thu May  6 00:48:39 PDT 2021; root:xnu-6153.141.33~1/RELEASE_X86_64 x86_64

```
Traceback (most recent call last):
  File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

carlosperate, 2021-06-10:
Thanks for the report @cbrown56!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `/Users/christinabrown/Library/Logs/mu/mu.log`

carlosperate, 2021-06-14:
Thanks the report @cbrown56!

I think this problem is already captured in issue https://github.com/mu-editor/mu/issues/1532, so to keep things tidy I'll merge this over there: https://github.com/mu-editor/mu/issues/1532#issuecomment-860614721

If you could attach your log file in that issue that would still be quite useful, thanks!

closed 2021-06-14 by carlosperate