mu-editor/mu #1632: Crash Report

rp10007, 2021-06-11:
[mu.log](https://github.com/mu-editor/mu/files/6635080/mu.log)

I have no idea whether this is relevant or not, but I did notice that I've got a copy of pgzero>=1.2.1 in local storage. Mu doesn't see it in it's own personal location, so it tries to download and goes boom. Endlessly repeatable.

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 5.4.0-73-generic #82-Ubuntu SMP Wed Apr 14 17:39:42 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 137, in download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 161, in run
    raise ex
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1
[mu.log](https://github.com/mu-editor/mu/files/6635099/mu.log)

```

rp10007, 2021-06-11:
Oh, and sorry, should have checked this earlier: b3 chugs for a while and then opens just fine, runs code.

carlosperate, 2021-07-05:
Oh interesting, so the issue has been introduced in beta.4?

We've just released beta.5, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

rp10007, 2021-07-11:
Thanks! beta5 (installed in its own venv) starts up and appears to run just fine. I will have to sort out my local environment before springing it on students...

closed 2021-07-11 by rp10007

carlosperate, 2021-07-12:
That's awesome, thanks for the update!