mu-editor/mu #1634: If sudo is used to `pip install mu`, running mu without sudo has issues

carlosperate, 2021-06-11:
There has been a couple of crash reports where it looks like the user might not have rights to download the wheels inside the system site packages (not 100% if this is confirmed, still waiting for responses in some of those crash reports).

I gave this a quick test in a Ubuntu 20.04 VM and in my tests Mu basically gets stuck at this point (it works otherwise when installed in a venv or without sudo):

```
$ mu-editor
2021-06-11 16:37:37,386 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-06-11 16:37:37,387 - root:266(run) INFO: uname_result(system='Linux', node='parallels-Parallels-Virtual-Platform', release='5.4.0-42-generic', version='#46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020', machine='x86_64', processor='x86_64')
2021-06-11 16:37:37,387 - root:267(run) INFO: Python path: ['/usr/local/bin', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/parallels/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages', '/home/parallels/.local/lib/python3.8/site-packages/IPython/extensions']
2021-06-11 16:37:37,387 - root:268(run) INFO: Language code: en_US
2021-06-11 16:37:37,387 - mu.settings:220(load) WARNING: No settings file found at /home/parallels/.local/share/mu/settings.json; skipping
2021-06-11 16:37:37,601 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-06-11 16:37:37,603 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-06-11 16:37:37,603 - mu.virtual_environment:542(ensure_path) INFO: Virtual Environment found at: /home/parallels/.local/share/mu/mu_venv-38-20210611-144638
2021-06-11 16:37:37,607 - mu.virtual_environment:555(ensure_interpreter) INFO: Interpreter found at: /home/parallels/.local/share/mu/mu_venv-38-20210611-144638/bin/python
2021-06-11 16:37:37,640 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 38
2021-06-11 16:37:37,641 - mu.virtual_environment:585(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-06-11 16:37:37,645 - mu.virtual_environment:611(ensure_pip) INFO: Pip found at: /home/parallels/.local/share/mu/mu_venv-38-20210611-144638/bin/pip
2021-06-11 16:37:37,645 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-06-11 16:37:37,716 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-11 16:37:37,716 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: flask
2021-06-11 16:37:37,939 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-11 16:37:37,940 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-06-11 16:37:37,962 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-11 16:37:37,963 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-06-11 16:37:38,013 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-11 16:37:38,013 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: adafruit_board_toolkit
2021-06-11 16:37:38,036 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-11 16:37:38,036 - mu.virtual_environment:496(ensure_and_create) INFO: Valid virtual environment found at /home/parallels/.local/share/mu/mu_venv-38-20210611-144638
2021-06-11 16:37:38,036 - mu.settings:169(save) DEBUG: Saving to /home/parallels/.local/share/mu/venv.json
2021-06-11 16:37:38,042 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x7f10f9be5ca0>; Splash Handler; <SplashLogHandler (INFO)>
```

If we want to overcome this we might want to check if the process running Mu has got write access to the wheels directory and if not we could have a fallback to download them in a folder in the same user directory where we create the user venvs?

Also, at this point I am not too sure what Linux distributions or versions require `sudo` to `pip install` without a virtual environment. In many experience this was necessary in a lot of the Linux environments I've worked with in the past, but when I tried it today in the Ubuntu 20.04 VM it was able to install Mu sudo-less without issues.

dhalbert, 2021-06-11:
> Also, at this point I am not too sure what Linux distributions or versions require `sudo` to `pip install` without a virtual environment. In many experience this was necessary in a lot of the Linux [...]

In my support experience, people used to do `sudo pip[3] install` because `pip3 install --user` was not the default, and the error it used to throw implied having to use `sudo` to be able to write system directories. But it is the default on most systems now. I always heartily disrecommend doing `sudo pip` because it can mess up the Python installs needed by the OS and its own programs.

`sudo pip` has also been popular on RPi because people are used to doing everything that way, and because controlling the pins, etc., is easiest to do with `sudo`. But using the correct group and/or adding udev permissions is a better idea.

https://stackoverflow.com/questions/29310688/sudo-pip-install-vs-pip-install-user
https://askubuntu.com/questions/802544/is-sudo-pip-install-still-a-broken-practice

carlosperate, 2021-06-15:
> In my support experience, people used to do sudo pip[3] install because pip3 install --user was not the default, and the error it used to throw implied having to use sudo to be able to write system directories. But it is the default on most systems now. 

Yeah, that was my guess, but I'm not too sure at this point what distros (and exactly what versions, specially for the LTS ones) have this by default.

> I always heartily disrecommend doing `sudo pip` because it can mess up the Python installs needed by the OS and its own programs.
>
> `sudo pip` has also been popular on RPi because people are used to [...]

Oh yeah, 100%, and we also recommend using a virtual environment on the documentation. Unfortunately based on the crash reports we receive there is plenty of people running `sudo pip install` with the system Python.

hroncok, 2022-04-19:
This also blocks us from packaging mu for Fedora. While `sudo pip install` is a bad idea, installing an app using the distro package manager is generally not.