mu-editor/mu #1635: Crash Report

axor, 2021-06-12:
**REMEMBER TO ATTACH YOUR LOG FILE**
[mu.log](https://github.com/mu-editor/mu/files/6642773/mu.log)

Version: 1.1.0.beta.4

Locale: en_US
Platform: Linux 5.8.0-55-generic #62~20.04.1-Ubuntu SMP Wed Jun 2 08:55:04 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/ghostbox/.local/lib/python3.8/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/ghostbox/.local/lib/python3.8/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/ghostbox/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/ghostbox/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 623, in create
    self.create_venv()
  File "/home/ghostbox/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/ghostbox/.local/share/mu/mu_venv-38-20210612-140936

```

carlosperate, 2021-07-05:
Thanks for the report and the log @axor!
We've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-08-12:
Hi @axor,

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@axor have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

axor, 2021-09-21:
Will try today at the end of the day.

axor, 2021-09-27:
Worked this time! Thank you.

carlosperate, 2021-09-27:
That's great, thanks for the update @axor!

closed 2021-09-27 by carlosperate