mu-editor/mu #1642: Failed install on windows 10 using Python 9

pviljoen2, 2021-06-14:
Traceback (most recent call last):
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

*** Note cannot find place to [upload]([mu.log](https://github.com/mu-editor/mu-editor.github.io/files/6470001/mu.log)) log file****

carlosperate, 2021-06-15:
Thanks for the report and the log @pviljoen2!

This are the interesting bits:

``` 
2021-05-12 19:08:04,868 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
```

```
2021-05-12 19:18:07,304 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\Pieter Viljoen\AppData\Local\python\mu\mu_venv-38-20210512-190755 as C:\Users\Pieter Viljoen\AppData\Local\python\mu\mu_venv-38-20210512-190755.FAILED
Traceback (most recent call last):
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Pieter Viljoen\\AppData\\Local\\python\\mu\\mu_venv-38-20210512-190755' -> 'C:\\Users\\Pieter Viljoen\\AppData\\Local\\python\\mu\\mu_venv-38-20210512-190755.FAILED'
```

carlosperate, 2021-06-15:
I've identified the two issues as https://github.com/mu-editor/mu/issues/1556 and https://github.com/mu-editor/mu/issues/1532, so to keep things tidy I've reference this crash report there and I'll close this issue to continue the conversation over there.

Thank you again for the report @pviljoen2!

closed 2021-06-15 by carlosperate