mu-editor/mu #1643: Crash Report

fruitpie999, 2021-06-15:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.14393 AMD64

```
Traceback (most recent call last):
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 522, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 573, in ensure_interpreter_version
    ok, output = self.run_subprocess(
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 373, in run_subprocess
    process = subprocess.run(
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid
```

carlosperate, 2021-06-15:
Thanks for the report @fruitpie999!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

fruitpie999, 2021-06-16:
[mu.log](https://github.com/mu-editor/mu/files/6659492/mu.log)
Don't have original log, but this is from today - same failure.

carlosperate, 2021-06-16:
Thanks for the log @fruitpie999!

Looks like we don't have much else in the log, the interesting bit is pretty much the provided stack trace:

```
2021-06-15 19:50:55,929 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x000001CF71ECD4C0>; Splash Handler; <SplashLogHandler (INFO)>
2021-06-15 19:51:02,961 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 522, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 573, in ensure_interpreter_version
    ok, output = self.run_subprocess(
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 373, in run_subprocess
    process = subprocess.run(
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] The parameter is incorrect
``` 

It looks like it got an issue at this point:
https://github.com/mu-editor/mu/blob/38f905a2fcb57b5438a39862b8823c31e74e0290/mu/virtual_environment.py#L562-L577

@fruitpie999 could you run this command in a command prompt (cmd.exe) and copy/paste the output here?
```
"C:\Users\breud\AppData\Local\python\mu\mu_venv-38-20210614-212859\scripts\python.exe" -c "import sys; print('%s%s' % sys.version_info[:2])"
```

fruitpie999, 2021-06-17:
C:\Users\breud>"C:\Users\breud\AppData\Local\python\mu\mu_venv-38-20210614-212859\scripts\python.exe" -c "import sys; print('%s%s' % sys.version_info[:2])"
38

carlosperate, 2021-07-05:
We've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

fruitpie999, 2021-07-05:
still getting errors with the new Beta version :-(

A couple trials from today with and without anti-virus, with and without running as Admin:

[mu.log](https://github.com/mu-editor/mu/files/6766159/mu.log)

carlosperate, 2021-08-12:
Thank you @fruitpie999,

Could you try running this command and paste the result here?

```
"C:\Users\breud\AppData\Local\python\mu\mu_venv-38-20210705-123423\scripts\python.exe" -c "import pgzero"
```

As a side note, this looks similar to this other crash report: https://github.com/mu-editor/mu/issues/1626.

fruitpie999, 2021-08-15:
Ran the above command ("C:\Users\breud\AppData\Local\python\mu\mu_venv-38-20210705-123423\scripts\python.exe" -c "import pgzero") and there were no results (blank).

carlosperate, 2021-08-16:
Thanks!

Okay, so we can more or less confirm that the venv is there and the import works, as no errors were printed.

That was command that Mu fails to run, but based on the error the problem arises before the command gets a chance to be run.

```
2021-07-05 15:37:11,468 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\breud\\AppData\\Local\\python\\mu\\mu_venv-38-20210705-123423\\scripts\\python.exe', '-c', 'import pgzero') with kwargs {}
2021-07-05 15:37:18,531 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 742, in ensure_key_modules
    ok, output = self.run_subprocess(
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 430, in run_subprocess
    process = subprocess.run(
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] The parameter is incorrect
```

This could be an issue with the Python version we ship with Mu. Could you try the following commands and copy/paste the output?

```
"C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\python.exe" --version
```

```
"C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import os; os.dup(1)"
```

```
"import os, msvcrt; print(msvcrt.get_osfhandle(0)); os.set_handle_inheritable(msvcrt.get_osfhandle(0), False)"
```

(the last two from https://bugs.python.org/issue37549)

fruitpie999, 2021-08-22:
C:\WINDOWS\system32>"C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\python.exe" --version
Python 3.8.5

C:\WINDOWS\system32>"C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import os; os.dup(1)"

C:\WINDOWS\system32>"import os, msvcrt; print(msvcrt.get_osfhandle(0)); os.set_handle_inheritable(msvcrt.get_osfhandle(0), False)"
'"import os, msvcrt; print(msvcrt.get_osfhandle(0)); os.set_handle_inheritable(msvcrt.get_osfhandle(0), False)"' is not recognized as an internal or external command,
operable program or batch file.

carlosperate, 2021-09-27:
Hi @fruitpie999,

That's for trying those commands and sorry for the late reply.

Could you try this command as well? (I missed something in the last command I wrote down):
```
"C:\Users\breud\AppData\Local\Programs\Mu Editor\Python\python.exe" -c  "import os, msvcrt; print(msvcrt.get_osfhandle(0)); os.set_handle_inheritable(msvcrt.get_osfhandle(0), False)"
```

fruitpie999, 2021-10-01:
I am no longer using the original computer, but have a new Win 10 system with the exact same problem.
The above command gives me 84

carlosperate, 2021-12-17:
Thank you @fruitpie999, I know it's been a while, but would be able to try a test installer to see if it resolves your issue?

Would you be able to try the installer at the bottom of this page, in the "Artifacts" section? (you need to be logged in GitHub for the link to the installer to appear)
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If it doesn't work, could you send us a fresh copy of the log? that would help us see if anything has changed with this installer. You should be able to find it in `%LOCALAPPDATA%\python\mu\Logs`