mu-editor/mu #1645: Crash Report

donxw, 2021-06-15:
[mu.log](https://github.com/mu-editor/mu/files/6659138/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\willi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\willi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\willi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\willi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\willi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-06-20:
Thanks for the report and the logs @donxw!

Looking at the log I think it might be this issue: https://github.com/mu-editor/mu/issues/1421#issuecomment-823662098

Did you have a previous beta of Mu installed before trying beta.4?
Could you try uninstalling Mu and then installing again beta.4? Hopefully that should remove the old wheels that can be causing this problem.

carlosperate, 2021-08-04:
Thanks for the report @donxw!
I'll close this specific issue as a being covered in https://github.com/mu-editor/mu/issues/1421.

If you could have a look at the release mentioned in this issue response and try the new Mu installer that would be really useful for us to confirm that it is fixed: https://github.com/mu-editor/mu/issues/1421#issuecomment-892776083

Thanks!

closed 2021-08-04 by carlosperate