mu-editor/mu #1646: Crash Report

Tozaki31, 2021-06-16:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: ja_JP
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-08-12:
Hi @Tozaki31 

Are you able to attach the log? File should be in `C:\Users\tozak\AppData\Local\python\mu\Logs\mu.log`.

Would you also be able to check if beta.5 has resolved the issue let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate