mu-editor/mu #1647: Crash Report

RayHamilton, 2021-06-16:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Ray\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Ray\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Ray\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Ray\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Ray\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```
No mu.log to attach (the directory was empty)

carlosperate, 2021-06-16:
Thanks for the report @RayHamilton!

Could you check again for the logs in this directory? Hopefully there is a log file by now.
```
C:\Users\Ray\AppData\Local\python\mu\Logs
```

RayHamilton, 2021-06-17:
Yes, there is one there now and hopefully attached! (It has been difficult
to actually find the file using windows explorer - some folders must be
hidden?)
I have tried installing again, actually logged on as Admin and had a
similar issue. I have tried installing on my Linux/Manjaro laptop and had
an issue with dependencies, but that is another issue!

carlosperate, 2021-06-17:
Thanks @RayHamilton!
We didn't get any logs on GitHub, if you tried to attach a file to an email unfortunately that doesn't get posted. If you could go into the GitHub issue page and add it there that would be quite useful, thanks!
Link: https://github.com/mu-editor/mu/issues/1647

carlosperate, 2021-08-12:
Hi @Timur-ccnp,

Are you able to attach the log via the GitHu website? File should be in `C:\Users\Ray\AppData\Local\python\mu\Logs`.

Would you also be able to check if beta.5 has resolved the issue let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@RayHamilton have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

If you could also attach the log (located in `C:\Users\Ray\AppData\Local\python\mu\Logs`) that would really help us investigate this issue.

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate