mu-editor/mu #1649: Crash Report

F3lixitas, 2021-06-18:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: fr_FR
Platform: Windows 7 6.1.7601 AMD64

```
Traceback (most recent call last):
  File "C:\Users\DEV\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\microbit.py", line 291, in flash
    path_to_microbit, port, board_id = self.find_microbit()
  File "C:\Users\DEV\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\microbit.py", line 234, in find_microbit
    board_id = int(serial_number[:4], 16)
ValueError: invalid literal for int() with base 16: 'AB0L'

```
[mu.log](https://github.com/mu-editor/mu/files/6675174/mu.log)

F3lixitas, 2021-06-18:
I don't know if it helps but the error occured when I had both an Arduino Nano and a Micro:Bit connected to the computer. When I unplugg the Arduino, Mu doesn't crash anymore.

carlosperate, 2021-06-20:
Thanks for the report and the log!
Interesting error, looking at the code the micro:bit mode should have not detected the board unless it was using the DAPLink USB VID and PID. What Arduino board were you using?

F3lixitas, 2021-06-21:
It was a regular Arduino Nano version 3.0
VID : 0403
PID : 6001

One detail I forgot, was that with the Arduino connected, Mu wants to switch to ESP Micropython mode on startup (wich I cancelled).

carlosperate, 2021-06-23:
Thanks @F3lixitas.
If you have only the Arduino plugged and try to flash do you get the same issue?
Could you attach here the details.txt file from the micro:bit?

F3lixitas, 2021-06-28:
It does crash even if only the Arduino is plugged in. 
Here's the detail.txt of my microbit device :
[DETAILS.TXT](https://github.com/mu-editor/mu/files/6723935/DETAILS.TXT)

F3lixitas, 2021-06-28:
I believe the issue comes from the software detecting the Arduino Uno as a flashable device. It is maybe trying to flash it instead of the microbit.
When I go into ESP MicroPython mode, it detects the Arduino :
[image]
COM98 is my Arduino's port.
Mu doesn't detect my microbit when it's connected :
[image]
Because of uflash this is not a big problem but maybe Mu replaces the missing device with the detected Arduino.